sudomesh/disaster-radio #11: p2p

samhavens, 2018-04-01:
Has there been consideration of trying to directly connect clients? An in-exhaustive comparison of potential pros and cons:

### Pros

1. Can extend the network. A U-shaped topology could become a disk if enough users were in the middle of the U.
2. Resiliency. One, many, or even all of the DR nodes could be compromised, but users could still share information.
3. Faster map loading. If a user has no maps cached, but has 5 users around them with cached maps, it is possible to parallelize the loading a la torrents. 

### Cons

1. Increased complexity
2. Potential battery drain

#### Implementations

The peer-to-peer communication could be implemented via one or more of:

* Bluetooth
* [Sound](https://github.com/robertrypula/AudioNetwork)
* IR - some phones come with both transmitters and receivers for IR, and it's also possible to build hardware specifically for this purpose

_Again, apologies if this is out of scope, has already been discussed, or is being asked in the wrong place_

samuk, 2018-09-27:
There's been some work on peer to peer messaging in other projects. Notably

http://www.servalproject.org/

https://briarproject.org/

samuk, 2020-02-28:
Think this could be achieved by hosting https://manyver.se/ or other apps on the SD card, and serving the files over Wifi. 

Related to https://github.com/sudomesh/disaster-radio/issues/59

closed 2020-02-28 by samuk