sudomesh/disaster-radio #13: Investigate supporting LibraryBox

lindner, 2018-07-09:
If possible it would be awesome to support [LibraryBox](https://librarybox.us/) or something like it.  

Then we could install Disaster Radios on [Little Free Libraries](https://littlefreelibrary.org/) everywhere.

randomodbuild, 2020-02-25:
Hey, I did something with a Little Free Library recently. It is not really off the grid, but I was thinking of making it as such. For a school project we made a Little Free Library, but we wanted to spice it up a bit. So I added an old android phone with Freedompop sim card (free 200mb) with a timer. Anytime from 6 in the morning to 9 at night if the phone detects motion (somebody closing the door, which shakes the accelerometer) then it will delay 10 minutes, delete the old photo in the folder and then take a photo. This will get detected by drive sync, and uploaded to a google drive account. Then, a wordpress website (littlecataloglfl.tk) has a drive plugin which takes the files in a drive folder and arranges them on the website. So people can go to the website to check which books are in stock before going. During the development, I really wanted to use LoRa because it wouldn't require a cellular/wifi connection but the free 200mb is the best we are gonna get. So maybe we could include something like this, but I don't know how well LoRa would send a 1mb image file 1-3 times a day. I would be willing to install it on our library for testing.

samuk, 2020-04-29:
I believe serving files is now working: https://github.com/sudomesh/disaster-radio/issues/59 please reopen if not

closed 2020-04-29 by samuk