sudomesh/disaster-radio #14: latest Makefile deleting all system files

paidforby, 2018-07-21:
On an almost fresh install of debian 9, following the instructions in the README, running `make` once, produces the error,
```
/bin/sh: 1: cannot create /tmp/mkESP/firmware_d1_mini: Is a directory
make: *** /tmp/mkESP/firmware_d1_mini: Is a directory.  Stop.
```
On trying `make` again, terminal blows up with stuff like,
```
rm: cannot remove '/lib/modules/4.9.0-7-amd64/kernel/sound/drivers/pcsp/snd-pcsp.ko': Permission denied
rm: cannot remove '/lib/modules/4.9.0-7-amd64/kernel/sound/drivers/opl3/snd-opl3-synth.ko': Permission denied
rm: cannot remove '/lib/modules/4.9.0-7-amd64/kernel/sound/drivers/opl3/snd-opl3-lib.ko': Permission denied
```
Further investigation, shows that all files that my user has access to have been deleted. Computer still works, but I'll have to reinstall debian to get it back to normal.

This seems related to Makefile update in https://github.com/sudomesh/disaster-radio/commit/3a3424fcd9a0c308aa9b8b6330a06947c2419b93  

After, `git checkout 6dcf338a3bb1484b125deb0c188c793183b72ad0` the last commit with old Makefile, `make` works as expected.

@Juul Is there a good reason for why we updated to the latest version? The old one worked ok (not great, but ok). Or is there a step I'm missing in setting up my computer for make?

Juul, 2018-07-21:
Fuck! That sucks :( That did not happen for me. I updated it because of issues consistently flashing the new non-USB boards.

Looks like this is the offending line:

```
$(shell rm -rf $(BUILD_DIR)/*)
```

:(

paidforby, 2018-07-21:
yeah, that line seems a likely culprit. I guess the solution would be to explicitly declare `BUILD_DIR` in the config.mk. Hesitant to try again, since I just reimaged my machine, but I guess there's no better time.

Juul, 2018-07-21:
Well we should change that line! That's a HORRIBLE line and should never
have been written in the first place!

noffle, 2018-07-21:
I like to add -i to my scripted rm -rf commands involving environment variables.

paidforby, 2018-07-23:
Found the issue, there were spaces after the BOARD and FLASH_DEF in setting.mk due to the addition of comments. Fixed in https://github.com/sudomesh/disaster-radio/commit/baafd262a999e069abe0395a2256f313c83c169d. I also removed the Makefile and added it to be retrieved by the fetch_deps script so its easier to tell what version we are using and to update it when necessary.

commit baafd262 (2018-07-23): remove spaces fixes #14 and fetch Makefile with other deps
closed 2018-07-23 by paidforby