sudomesh/disaster-radio #16: maptiles loading slowly or not at all

paidforby, 2018-08-14:
When connected to a disaster radio SSID and going to http://192.168.4.1/maps/ the page might load once the first time, but on a second reload is the map tiles load either slowly or not at all. Issue appears to be a combination of not caching the tiles at all and too many simultaneous http requests. 

## workarounds
* cache map tiles for a sane amount of time (1 week?) by setting cache control in header, https://github.com/me-no-dev/ESPAsyncWebServer/issues/181
* after loading page to your browser once, scroll slowly so it only tries to load a few new tiles at a time

jnny, 2018-09-04:
did this commit fix the loading time issue?

samuk, 2020-02-28:
Think this is resolved, please reopen if not

closed 2020-02-28 by samuk