sudomesh/disaster-radio #18: Cannot exchange messages across two clients connected to same node

benhylau, 2018-08-22:
Got a pair of dev boards from dweb, but could not get them to talk to one another on the 192.16.8.1.4 chat. Build firmware and web app from `master` following the instructions here, got web app running again, but still cannot get two laptop clients connected to the chat to exchange messages. Each client sees their own view of the chat.

The software build toolchain is pretty solid though :)

paidforby, 2018-10-12:
Sorry, just noticed this issue. to clients on the same node do not see their eachother's messages because they are not getting pushed to the websocket. There does exist a hack to enable very similar functionality, if you enter `$' in the chat app after logging in, echoing will be toggled (or you can [flip this bit](https://github.com/sudomesh/disaster-radio/blob/master/firmware/firmware.ino#L63) in the firmware and recompile it). Then messages will be echo'd to the websocket, so the client who posts the message will appear to have sent two messages, but other clients on the node will only see the echo. Recent versions of firmware make this and other functionailty more visible by sending debugging output to the websocket, like [this](https://github.com/sudomesh/disaster-radio/blob/master/firmware/firmware.ino#L345), probably should add a line that tells you how to enable echo if it is off.

benhylau, 2018-10-13:
@stigtsp and I got two devices to talk to each other as of a week ago. They were behaving a little unpredictably at times, but they were also running different version of firmware. We will flash latest and try again. Thanks for the tip about echo.

paidforby, 2018-10-13:
Yes, I definitely recommend flashing off of latest, I'm sure there are still some bugs, currently working on a complete revamp of the firmware using our simulator, https://github.com/sudomesh/disaster-radio-simulator.

lewisxhe, 2019-12-03:
I also cannot exchange information between two nodes. I use esp32 as the node, as shown in the figure.
[image]

lewisxhe, 2019-12-03:
I have understood that I will try to adapt to TTGO Twatch

paidforby, 2019-12-04:
It looks like you are receiving the chat message. The debugging message "I am not the next hop, packet dropped" is a false alarm, the packet sent by the demo chat app is not a routed packet, it is a broadcast packet with a destination address of `FF:FF:FF:FF:FF:FF`. Since this doesn't match the receiving node's address and does not match any entry in it's routing table, from a routing perspective, it rightly drops the packet. As I write more of the routing logic for chat messages, I may change the handling of broadcast packets, but currently that is the way it works. 

To debug the sending and receiving of chat messages, open the chat app of the receiving node and see if you are seeing the messages from the sending node. Based on your second comment, I'm guessing you figured that out.

samuk, 2020-02-13:
@lewisxhe did you get anywhere with the TTGO watch? Do you need further help or can this issue be closed?

lewisxhe, 2020-02-14:
I have no problem.@samuk

samuk, 2020-02-14:
OK thanks for letting me know. Did you get anywhere with the watch?

Would be good to add it to this page if you got it working https://github.com/sudomesh/disaster-radio/wiki/Devices--&-Hardware

closed 2020-02-14 by samuk

lewisxhe, 2020-02-15:
Sorry, I did n’t finish this@samuk