sudomesh/disaster-radio #19: add a compass

nicopace, 2018-10-07:
can be used to help you walk from where you are to a location that you tapped based on the map app.
you can use this as an example: https://github.com/lamplightdev/compass

samuk, 2020-05-13:
I guess this could be useful on GPS enabled boards?

nicopace, 2020-05-13:
The ones that are GPS enabled are the phones... but then, in order for the GPS of the phones to be used, they will need HTTPS, and that would require a TLS enabled web server like this one: https://github.com/fhessel/esp32_https_server
Still, if you develop native apps, this could work :)

paidforby, 2020-05-13:
I see what you are @nicopace suggesting. This could be fairly easy to add to the web app. If/when someone works on a revision of the web app, they can attack this issue. @samuk there could also be a compass displayed on OLED screen, meshtastic does something like this on GPS-enabled ESP32 dev boards, see the video on their homepage https://www.meshtastic.org/