sudomesh/disaster-radio #22: 2 clients not talking

briangraphted, 2018-12-01:
Hi - Not sure this is the proper venue but here goes.

I have 2 huzzah esp8266's each with their own rfm95 lora running on the 915 frequency. I've been at this for a while (meaning trying to get these to talk) but have hit a dead end. Here's my bench top setup:
- 2 mac laptops, each with a usb/ftdi cord connecting directly to an esp8266 for power
- each esp8266 is connected (per the pin hook up guide) to the rfm95 lora
- each esp8266 is flashed (make flash and make flash_fs) with the firmware and the web app. NOTE: I do *not* have an SD card hooked up, nor a solar panel.
- on each laptop I open a serial terminal 
- manually reset (push on board button) each esp8266 and watch the proper message in the serial monitor (example message below). 
- on each laptop I connect to the appropriate esp8266 AP (2 laptops, 2 esp8266, 2 wifi access points). The access points read: disaster.radio 18fe34d45dd4 and disaster.radio 18fe34d40f15. 
 - on each laptop I open http://disaster.local
 - on each laptop I put in a name (nick) like "node1" then "node2" (on the 2nd laptop)

Observations
The web app loads. The debug messages *say* the lora is initialized but I'm suspicious. I see the bootstrap routes being sent (i.e.: "Sending: 011018fe34d45dd4ffffffffffff2272) - meaning that ap one is broadcasting widely / to no specific address. But I never see a corresponding receive from "node2" either in the serial monitor nor the web app. I've tried to debug sendMessage on the websocket app but to no avail. ... I have this itch at the back of my mind that I'm missing something *super* basic like the proper URL in the browser or something. 

Help would be appreciated!
-brian

<snip from serial monitor>
enabling SD card

Waiting for SD card to initialise...SD Card Not Found!
ok
The file exists!
Size=269
<!DOCTYPE html>
<html>
  <head>
    <title>Disaster Radio</title>
    <link rel="stylesheet" href="build/index.css">
  </head>
  <body>
    <h1>Disaster Radio</h1>
    <div id="container">
      
    </div>
    <script src="build/bundle.js"></script>
  </body>
</html>

LoRa init succeeded.
local address: 18fe34d45dd4
add 1
aid 1
station: 44:2a:60:f3:45:96 join, AID = 1
transmitting routes

ttl: 1
length: 16
source: 18fe34d45dd4
destination: ffffffffffff
sequence: 00
type: r
data: 
Sending: 011018fe34d45dd4ffffffffffff0072
routes transmittedtransmitting routes

paidforby, 2018-12-02:
Hi @briangraphted ! Cool you trying to get the disaster radio firmware working. I can't say I've tried running our firmware on the huzzah boards, how is the esp8266 is connected to the lora transceiver? On our hardware the pinouts are as follows,

NodeMCU/ESP8266 | LoRa Transceiver
----------------|--------
D1/GPIO5/OUT | RESET 
D2/GPIO4/INT | DIO0
D5/GPIO14/SCK | SCK
D6/GPIO12/MISO | MISO
D7/GPIO13/MOSI | MOSI
D8/GPIO15/CS | NSS

My guess is that the DIO0 pin of the transceiver is wired to a different pin of the esp8266, this would allow lora to initialize, but prevent the onReceive interrupt from being triggered.

briangraphted, 2018-12-03:
Hi -
Thanks for the fast reply. No luck yet but still poking at it. I'll reply here when I've got it worked out.
thanks.

briangraphted, 2019-02-04:
Sorry for long delay. These pin connections work when hooking up a Wemos D1 Mini (esp8266) and a Adafruit RFM95W LoRa Radio Transceiver Breakout - 868 or 915 MHz 

D1 - lora
GND - GND
3v3 - VIN
D8 - CS
D7 - MOSI
D6 - MISO
D5 - SCK
D1 - G0

closed 2019-02-04 by briangraphted