sudomesh/disaster-radio #23: Help with gerber files & finding suppliers

krisradio, 2019-02-20:
o/ all,
I find this project a couple days ago and really want to put together my own disaster radio. Potentially bring the system to a few mutual aid organizations active in disaster response. 

Honestly I'm pretty new to hardware so the gerber files alone are confusing me. If I want to buy some of the v4 boards, should I zip up all the files in hardware/board_v4? Or just the three files in /hardware/board_v4/gerber? 

Second question, does anyone know a supplier that can ship all/most of the parts on the build list? Every link is to a different source. I've been trying to find one storefront with everything to save on shipping/complexity, but no luck so far. Would love to hear where y'all got your parts. 

Thanks for reading, 
Kris with a K

(p.s. sorry if an issue is the wrong way to ask this, but I couldn't find any general/personal contact details)

krisradio, 2019-02-22:
Hi all, 
trying to put together a shopping list and I had a couple questions about the power supply. 
The build list links to a pair of UltraFire 3.7V 3600mAh batteries (for $4, not sure where you'll ever find them at that price), but it seems to me that AA batteries would be harder to connect to the board than wired battery packs, like these [2500mAh](https://www.adafruit.com/product/328) or [4400 mAh](https://www.adafruit.com/product/354) packs from AdaFruit. 
My question is, do either 2500 or 4400 mAh packs work for disaster.radio (with some tinkering presumably)? 

Or do I need to get AA batteries, along with some spares & chargers [from amazon](https://www.amazon.com/6000mAh-Battery-Rechargeable-Batteries-Flashlight/dp/B07F29JZMR/)? If so, I need an AA battery enclosure too, right? 

For the antenna it looks like [AdaFruit has a compatible option](https://www.adafruit.com/product/3340). 

For the WeMos D1, looks like the same supplier linked for the LoRa chip has a [slightly older version](https://www.aliexpress.com/store/product/Mini-D1-Mini-NodeMcu-Lua-WIFI-ESP8266-based-on-the-ESP-12F-module-development-board/1795152_32817642418.html). Pinout looks the exact same so I think/hope that one will work. 

For the solar panel, I'll probably just go with the [linked one](https://www.ebay.com/i/121104191012?chn=ps&dispItem=1), but if anyone knows how reliable that supplier is lmk (I can see some damage in these pics). 

Also, I still have no idea what to do with these gerber files. For reference I'm an electrical engineering student now, but more of a computer science background. I'm just getting into this world of soldering, power supplies, etc etc. and would really appreciate a helping hand. I'm also an active volunteer in disaster response and really want to bring this project to a few organizations I work with. 

In solidarity, Kris with a K

krisradio, 2019-02-23:
Hello again, just ran the numbers on ordering these custom .pcb's through seeed according to the order placed [March 20th, 2018](https://github.com/sudomesh/disaster-radio/tree/master/hardware/orders/order3_march_20th_2018). Unfortunately the cost for the minimum order size might be breaking my bank at $102.24. Honestly that's so far off from the $5 estimate that I'm a little annoyed this is billed as a $40 project. 

On another note I was looking at options for the panel/batteries and saw this [adafruit guide](https://learn.adafruit.com/usb-dc-and-solar-lipoly-charger). Although that too would raise the price, the article says their lineup eliminates the need for a buck converter, so that would take one item off the list. Plus, the adafruit solar panels look a little more reliable/beefy than the ones linked in the build guide. Thoughts?

krisradio, 2019-03-22:
Been doing some more research into gerber files and comparing the march 20th order to the v4 files. Gerber is a thick format but I'm pretty sure the march20th order, with absolute drill coordinates plotted over (-6, -3) inches away from the origin, would be wildly off the 33x26mm size of the board. 

Meanwhile, the v4 files have two drill layers but only one copper layer. So, both of these files are very confusing to me, and I'm not surprised PCB Way said they couldn't print them. Would love some guidance on this issue.

samuk, 2020-02-13:
Do you still need support with this? Or would one of these dev boards meet your needs?

https://github.com/sudomesh/disaster-radio/wiki/Devices--&-Hardware

samuk, 2020-02-28:
No response to this old issue

closed 2020-02-28 by samuk