sudomesh/disaster-radio #24: Can I help organize a "v4" todo list?

somlor, 2019-03-19:
Hi, I just found this project and I'm super interested in setting one up in my community (currently Asheville, NC).

I noticed under [disaster.radio/build](https://disaster.radio/build):
> Our first prototypes had a few bugs and we're currently working on a fixed design. 

And via the mailing list from [November 2018](https://sudoroom.org/pipermail/disasterradio/2018-November/000103.html):
> The new board is located in the repo here (https://github.com/sudomesh/disaster-radio/tree/master/hardware/board_v4)

Would it help to have an organized public todo list of all remaining tasks needed to get from wherever things are at now to a goal of, say: 
  - New, fully-tested and functional ES32-based "v4" design
  - Updated [GitHub wiki](https://github.com/sudomesh/disaster-radio/wiki)
  - Updated [Learn/Build sections of website](https://github.com/sudomesh/disaster-radio-website)

If so, please let me know if I can help organize and/or maintain something like this.

krisradio, 2019-03-20:
Hey Somlor, good to see some interest in the project, I've been trying to build my own box but haven't gotten any feedback from the devteam so far. Keep us updated on your progress. I hope to have some prototypes within the next three months and I'm happy to help with documentation/todo any way I can.

samuk, 2019-06-18:
I'd love to see this project continue.

It looks like there has been some work done on the Esp32 branch 
https://github.com/sudomesh/disaster-radio/tree/esp32

This looks like promising hardware with on-board SD 
https://www.aliexpress.com/item/TTGO-LoRa32-V2-1-1-6-version-868-915Mhz-ESP32-LoRa-OLED-0-96-Inch-SD/32872537745.html

I'm waiting for a board, but can't currently get the Esp32 branch to compile.

krisradio, 2019-06-23:
That's an interesting piece of hardware @samuk. Seems like utilizing it over the WeMos dev board and current LoRa chipset would require almost completely refactoring the hardware at least, potentially software too. But it looks like a pretty nimble piece of hardware, if it could replace both those and reduce the construction time/cost, it could be worth it. 

I'm still really intrigued by this project as well. Right now I am still confused by the lora_nodemcu.drl file in the [march 20th order](https://github.com/sudomesh/disaster-radio/tree/master/hardware/orders/order3_march_20th_2018). The drill holes are not aligned to the pins, the board, anything. 

I am switching gears towards trying my hand at running [althea firmware](https://github.com/althea-net/althea-firmware) on commodity routers. If I can get a little experience with that I might be able to install althea on disaster radio by the time of the next release. Then I could spin up a mesh net with fully decentralized, p2p, real-world incentives. A lofty goal but I'm not the only one dreaming of it.

samuk, 2019-06-23:
I'm very interested in getting the software adapted for this off the shelf hardware. I think if people can just order a $17 device off Aliexpress it's going to appeal to a greater number of people.

 I lack the skills to do it personally, but I can try and get other people excited about the idea.

krisradio, 2019-06-26:
Yeah it has the LoRa, battery, and storage, but does it have a processor architecture/speed compatible with disaster.radio or another mesh networking protocol? That's the question I can't find an answer to on the store listing.

paidforby, 2019-11-05:
FYI https://github.com/sudomesh/disaster-radio/pull/25
I'm revisiting parts of this project. I can't speak much to the hardware development. I am mostly interested in getting firmware and routing protocol working on the ESP32 TTGO board suggested by @samuk so that other people can possibly work on hardware prototypes and real-world deployments. 

I will create check list on the [Github wiki homepage](https://github.com/sudomesh/disaster-radio/wiki) so y'all can keep track of my progress, or others can add their goals/progress.

closed 2020-02-13 by samuk