sudomesh/disaster-radio #26: RSS feed of messages?

samuk, 2019-11-07:
Would it be easy or complicated to add a RSS feed that exposes all the messages?

A user could then use a reader https://search.f-droid.org/?q=rss&lang=en

To get notifications/ alarms when a new message is received.

paidforby, 2019-11-13:
Cool idea! I'm not sure how difficult it would be. I know very little about RSS feeds. By "exposes all the messages" do you mean seeing an entire history of messages sent and received on the node? I guess the first step would be revisiting the `storeMessage` and `dumpLog` in [main.ino](https://github.com/sudomesh/disaster-radio/blob/esp32/firmware/main.ino#L95). I haven't looked at/used these in a while and they probably could use an update/sync with latest code. Once you have a log of all received messages, you would need to create a properly formatted RSS feed xml file, like [this example](https://en.wikipedia.org/wiki/RSS#Example). Then the generated xml file would need to placed inside the `/web/static` directory. The real challenge would be doing all the necessary string manipulation to generate the xml since there's no nice tools like sed/awk on the ESP.

It's doable and would take a decent amount of work. It would also be a good reason to revisit the logging of messages. side note: there was some work done on an SSB gateway, which could serve a similar purpose, find that on git-ssb [here](https://git.scuttlebot.io/%25SyAU1pa6g6yYcELjTrBG4JY41vjuP6HSvAo%2BmsrP0Sg%3D.sha256).

samuk, 2020-05-13:
This would be kinda interesting still, largely superseded by the Android app that solves this use case.

closed 2020-05-13 by samuk