sudomesh/disaster-radio #30: Default IP 192.168.4.1 - add to instructions

samuk, 2019-12-05:
I was easily able to flash the new release to a TTGO board, I couldn't find the default IP though.

I worked out it was http://192.168.4.1/ but could be worth adding to the instructions.

In the longer term, it would be good to add DNS and a captive portal/ redirect so users finding an open device end up on the right page. It looks like it's possible eg https://github.com/dproldan/Esp32AutoCamera 

Waiting on a second board so that I can test properly.

paidforby, 2019-12-06:
Glad to hear you were able to easily flash the firmware. 

You make a good point about that IP address. I will note that it is listed in the [firmware directory's README](https://github.com/sudomesh/disaster-radio/tree/master/firmware), but that is a little hard to find if you aren't digging through the repo. I will add it to the 0.1.0 release instructions and to the main README.

re: DNS
Running DNS on these ESPs is inconsistent (in my experience) some devices don't play nice with it trying to act like a DNS server. I haven't tested it in a while, maybe it has gotten better.

re: captive portal
I have an aversion to captive portals. I generally think they have a negative connotation (think "free" wifi at hotels and airports). Though, you are correct, it is very possible and very easy. I even wrote an adaptation of that example for an old gig [here](https://github.com/shapeable/esp8266_probe_sniffer/blob/master/src/asyncCaptivePortal.ino). I'm not against it being an option, just would rather not have it be the default.

I'm actually more interested in the option of having the disaster.radio joining your home wifi network and then being accessible at an IP address (or DNS name) on your home LAN. I realize this brings in some uncontrolled variables, but I think it makes more sense. This way disaster.radio just another "IOT" service on your home network. I think this would also make your RSS feed idea (#26) practical and useful.

In the event that the disaster.radio loses it's connection to your home network, it would fall into "disaster.mode" and become an AP again.

paidforby, 2019-12-06:
release has been updated with information regarding usage of chat app https://github.com/sudomesh/disaster-radio/releases/tag/0.1.0
I'm planning a larger overhaul of the main repo's readme, I will update it according when I follow through with that plan

samuk, 2019-12-06:
Great, thanks.

I'll have a go at hacking the captive portal stuff and report back any success.

I got hold of my second board and did some successful tests over around 400m. I'll try further tomorrow.

closed 2019-12-06 by samuk

mofosyne, 2021-01-16:
From a user UX perspective of a random person stumbling across a disaster radio node, then a captive portal is best. Since the device is the only one the person is connected to.

However if its connected to a local wifi network as one amongst many devices... something like how octopi does local broadcast DNS via https://en.wikipedia.org/wiki/Multicast_DNS so you can access it like `disasterradio.local` would be quite convenient.

Pablo2048, 2021-01-16:
The mDNS is already used ( https://github.com/sudomesh/disaster-radio/blob/a90171204ca6aa40928c36d9eae6ce6c6aa49bb7/firmware/main.ino#L161 ), but remember that the mDNS `.local` resolution is not working on Android devices.