sudomesh/disaster-radio #34: Write chat history to Micro SD (where available)

samuk, 2019-12-28:
Where there is an SD card installed is it easy to log the chat history to SD card and load it when a new browser session is started?

paidforby, 2019-12-29:
Yes, it should be simple enough. I wrote logic for this some time ago, see it [here](https://github.com/sudomesh/disaster-radio/blob/master/firmware/esp32_ttgo/main.ino#L94). I haven't used it in quite a while, but plan on revisiting it soon.

samuk, 2019-12-29:
Great! So if I access http://192.168.1.4/log.txt I should be able to see it in a browser? (Not near a device to test at the moment)

paidforby, 2020-01-02:
Not quite because the `storeMessage` function is not utilized in the latest code. Even if it was implemented, you would not find it at http://192.168.1.4/log.txt because `log.txt` is being stored to the root of the SD card, but to be served up by the web app it needs to be stored in the `/static` folder. I'm not sure if there are any other catches, but this feature should be easy enough to implement assuming that `storeMessage` function still works.

samuk, 2020-02-01:
https://github.com/sudomesh/disaster-radio/blob/master/firmware/esp32/history/HistorySD.cpp

closed 2020-02-01 by samuk