sudomesh/disaster-radio #35: Regular node check in auto-post

samuk, 2019-12-28:
Would it be difficult to generate a auto-post from each node every day, or week, just advertising that the node is still live?

I realise you wouldn't want to do this too often from a network overhead point of view, but it would be good to know that nodes were still up every week or so.

paidforby, 2019-12-29:
Yes, this is already implemented to a degree. For a node to maintain it's routing table, it needs to ping it's neighbors to make sure they are still alive. This ping is sent by the `LL2.daemon()` that is called in the main loop. The interval between pings is set by the `LL2.setInterval(<time_in_ms>)` function in the main setup, see [these lines](https://github.com/sudomesh/disaster-radio/blob/master/firmware/esp32_ttgo/main.ino#L463). For development, I have the interval set to a very low 10s. In reality, you may only send this ping out every 10mins or every hour. 

Ideally, this interval would be set in a web app config gui, in which we could also print the routing table to see what all nodes are alive on a given network.

samuk, 2019-12-29:
Great!

The config bit sounds a bit involved, as a first step could the routing table just be printed below the chat app?

paidforby, 2020-01-02:
Yes. Something like that could/should be done. Right now (since I'm actively avoiding getting bogged down in developing the web app) it might be easier to create a command that can be entered into the chat app that prints the routing table. This already exists in the latest console interface, you can print the nodes current routing table using the command `lr -r`. I'll work on cooking up a stable release that includes the new console interface features. And I'll think about ways to make this routing table info available to the web app.

paidforby, 2020-01-16:
> The config bit sounds a bit involved, as a first step could the routing table just be printed below the chat app?

This is now a reality. 

[image]

I based it mostly off parts of the simulator that perform a similar function. If a node is added to the network, the table in the web app will update every time a routing packet is sent which is determined by `LL2.setInterval()`.

This is improvement I was waiting to make before releasing 0.1.1,  so keep an eye out. 

Eventual improvements, 
* create a separate interval for sharing information between the firmware and the web app (this will be more involved, but more useful for getting info like the node's mac address and hostname)
* make the active node list clickable, to easily "@" another node (i.e. send a routed packet to a specifc node)

closed 2020-01-17 by samuk