sudomesh/disaster-radio #37: Function returns pointer to a local

dwhall, 2020-01-04:
firmware/esp8266_wemos/routing.cpp:631
packet_received() returns a pointer to a local variable.

dwhall, 2020-01-04:
Sorry for the noise.

paidforby, 2020-01-05:
No noise. This actually brings up an important issue. The esp8266 code needs to be synced with the latest developments. I haven't touched that code in almost a year and have switched to developing on the esp32. We should either get that esp8266 code actually working, or we should stop claiming to support it. Thanks for bringing this up. I'm going to rename and reopen this issue.

samuk, 2020-05-13:
@paidforby Is this still on your to-do? Does it make sense to spend energy on it? Suggest just dropping support for esp8266 unless there's a compelling reason?

paidforby, 2020-05-13:
It does not. Pretty much everyone has appeared to move on to the ESP32. I'm considering ways to restructure the firmware folder in a more sensible way. When I do that, I'll remove the stale ESP8266 folder to avoid confusion.

closed 2020-05-13 by paidforby