sudomesh/disaster-radio #39: Website has a broken link to enclosure stl file

kellybyrd, 2020-01-14:
The [hardware page](https://disaster.radio/learn/hardware/) has a broken link to the enclosure STL.
The broken link is:
https://github.com/sudomesh/disaster-radio/blob/master/enclosure/disaster_radio_enclosure.stl

I think you want to link to this directory:
https://github.com/sudomesh/disaster-radio/tree/master/enclosure

paidforby, 2020-01-14:
It is actually meant to link to this file https://github.com/sudomesh/disaster-radio/blob/master/enclosure/old/disaster_radio_enclosure.stl but that was moved into the `old` directory. There are a lot of other updates that need to be made to the site. Thanks for pointing this out!

paidforby, 2020-03-21:
Fixed in recent update to website, https://github.com/sudomesh/disaster-radio-website/commit/530b47816da6810dd33e8dbe76ac1ad42afc8115

Took your suggestion and pointed to enclosure directory, see update to hardware page https://disaster.radio/learn/hardware/

closed 2020-03-21 by paidforby