sudomesh/disaster-radio #40: heltec_wifi_lora_32_V2 support

matthewrwright, 2020-01-15:
Hi - trying to compile for the heltec_wifi_lora_32_V2 board using platformio - compilation succeeds but I'm getting a lot of warnings about pins being overridden in code. 

`In file included from .pio/libdeps/heltec_wifi_lora_32_V2/LoRaLayer2/src/Layer1_LoRa.cpp:2:0:
.pio/libdeps/heltec_wifi_lora_32_V2/LoRaLayer2/src/Layer1.h:25:0: warning: "LORA_DEFAULT_RESET_PIN" redefined
 #define LORA_DEFAULT_RESET_PIN     23
 ^
In file included from .pio/libdeps/heltec_wifi_lora_32_V2/LoRaLayer2/src/Layer1_LoRa.cpp:1:0:
.pio/libdeps/heltec_wifi_lora_32_V2/LoRa_ID1167/src/LoRa.h:20:0: note: this is the location of the previous definition
 #define LORA_DEFAULT_RESET_PIN     9
 ^
`
Unsure if this is the source of my problem, but I get a warning that my LoRa radio isn't connected when I launch the serial console. 

`     ___              __                            ___
 ___/ (_)__ ___ ____ / /____ ____      _______ ____/ (_)__
/ _  / (_-</ _ `(_-</ __/ -_) __/ _   / __/ _ `/ _  / / _ \
\_,_/_/___/\_,_/___/\__/\__/_/   (_) /_/  \_,_/\_,_/_/\___/
v0.1.0
WARNING: LoRa tranceiver not found!
Local address of your node is a4cf1202ad5c

Sending: 0110a4cf1202ad5cffffffffffff0172
`

Any clues gratefully accepted!

paidforby, 2020-01-15:
Interesting. It looks like there is an unintentional conflict between a constants in the LoRaLayer2 and LoRa libraries. I haven't seen this issue when compiling with makeEspArduino for the LILY TTGO dev board. I'll change the names of the constants so there is no conflict.

However, I'm not sure that is your issue. I haven't been developing with the Heltec dev board, so I'm not sure what pins it has connected to its LoRa transceiver. Maybe there are examples or documentation the would provide hints. Once you figure out what pins it uses, you can manually set the pins by adding the following line in the `setup()` function of `main.ino` before calling `Layer1.init()`,
```
Layer1.setPins(int cs, int reset, int dio);
```
Good luck! Let me know what pins work for that board so we can document it somewhere.

paidforby, 2020-01-15:
Didn't mean to close issue. But I did update LoRaLayer2 with a partial fix. I renamed the pin constants, so there should no longer be a conflict if you pull the latest commit of LL2. I'll update platformio config to pull the latest.

matthewrwright, 2020-01-15:
Thanks! Pulled the latest commit and set the pins as-per the pinout diagram but no joy

https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/PinoutDiagram/WIFI_LoRa_32_V2.pdf

I'll keep plugging away and see if I can get it running.

paidforby, 2020-01-16:
Hmmm, I know there's some unreliable documentation out there for ESP32 pinouts. Maybe that diagram doesn't show the correct pinout? Or it could be disguising something, like maybe the OLED chip select needs to be pulled low? Wish I could be of more help.

I did find that Heltec has a whole repository of [source code and example code](https://github.com/HelTecAutomation/Heltec_ESP32). It actually looks like they have their own port of sandeep's arduino-LoRa,

https://github.com/HelTecAutomation/Heltec_ESP32/blob/master/src/lora/LoRa.h#L7

It's not clear what the difference is outside of a simplified header file. Perhaps there are some clues in that directory that will solve your problem. At the very least, you could try flashing some of their [LoRa code examples](https://github.com/HelTecAutomation/Heltec_ESP32/tree/master/examples/LoRa) to confirm that your board is working.

matthewrwright, 2020-01-19:
Thanks for the pointer - their sample OLED_LoRa_Sender works when compiled with Arduino, so now to find the differences!

matthewrwright, 2020-01-20:
I'm at a bit of a loss as to what's happening - I've even tried changing the code to use the Heltec version of the library. I haven't checked the code for pin reuse - I'm wondering if the calls to initialise the SDCard which this board doesn't have are setting a pin. Will look into it more over the next couple of days. 
I know you have a lot to look into, so appreciate your time on this.

matthewrwright, 2020-01-24:
Happy to report that the code from PR #44 works with additions to platformio.ini

```
[env:heltec]
platform = espressif32
board = heltec_wifi_lora_32_V2
framework = arduino
upload_port = /dev/ttyUSB0
monitor_port = /dev/ttyUSB0
monitor_speed = 115200
build_flags =
        -DHELTEC

```
and config.h

```
#ifdef HELTEC
#define LORA_IRQ DIO0
#define LORA_IO1 DIO1
#define LORA_IO2 DIO2
#define LORA_SCK SCK
#define LORA_MISO MISO
#define LORA_MOSI MOSI
#define LORA_RST RST_LoRa
#define LORA_CS SS
#define OLED_SDA SDA_OLED
#define OLED_SCL SCL_OLED
#define OLED_RST RST_OLED
#define OLED_WIDTH 128
#define OLED_HEIGHT 64
#endif
```

As it looks like #44 is going to be merged can we add a definition for Heltec boards to config.h afterwards? Happy to make a PR for changes afterwards or for you to make the changes after the refactor.

paidforby, 2020-01-24:
Very nice. That is great to hear. I can just add the lines you provided after merging the refactor, if that is OK with you. I have a few other boards to test, so I'd like to think about the best way to include all of them.

matthewrwright, 2020-01-24:
Perfect - thanks! Those pin definitions came from the paxcounter repo mentioned in the thread of #44 so certainly worth a look there for experimental support for various boards which can be confirmed as/when people test them.

matthewrwright, 2020-01-29:
Now #44 is merged, it looks like my changes to platformio.ini and config.h both work fine for the heltec. Now to get some range testing in!

paidforby, 2020-01-30:
Added your suggestions in this commit https://github.com/sudomesh/disaster-radio/commit/be1d6ff6c41a050ff14465e169ac644bb99e8a32 you'll have to confirm that I copied everything correctly and that it still works for you. (p.s. I also changed the name of the firmware directory to be more general since we now support more that just TTGO boards!)

closed 2020-01-30 by paidforby