sudomesh/disaster-radio #41: Low power repeater nodes

samuk, 2020-01-17:
I'm looking to put up some 'infrastructure' nodes in high places. These nodes will be solar and so power is important.

On the TTGO boards I measure around 160mW to peaks of 250mW.

It seems that these 'infrastructure' nodes do not need all the functionality of the webserver and Wifi.

So we could create a branch that;

* Does not use Wifi
* Does not include a webserver
* Runs at a [lower clock speed](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-cpu.h#L38)?

Juul, 2020-01-17:
Do the TTGO boards connect the receive interrupt pin from the LoRa module? If so, you could get even lower by putting the ESP32 into a sleep mode and only waking up when a packet is received.

I've been working on getting very low power on an Atmega chip recently and would be interested in playing around with this.

You might also want to run the ESP32 at lower voltage, though I'm not sure it can go that low.

matthewrwright, 2020-01-17:
Interesting idea - I live in a valley almost, so it would certainly be worthwhile to mount a couple up high on the hills to extend coverage. Having something with a reasonable battery life would cut the need for a large solar panel and could report it's voltage to the mesh when it got low and someone could change the battery out.

samuk, 2020-01-24:
Edited the title as Wifi/ Bluetooth being discussed on this ticket herehttps://github.com/sudomesh/disaster-radio/issues/45

samuk, 2020-02-08:
I had a play with the clock speed, reducing to 80Mhz seems to have no adverse effects. I tested in BT mode. This reduces the power consumption to ~70mA, with peaks of ~120mA on broadcast.

I just added 
```
setCpuFrequencyMhz(80); //Set CPU clock to 80MHz fo example
getCpuFrequencyMhz(); //Get CPU clock
```
To line 424 of main.ino

samuk, 2020-02-12:
https://github.com/sudomesh/disaster-radio/pull/54#pullrequestreview-356251424

closed 2020-02-12 by samuk