sudomesh/disaster-radio #42: Support FCC/ CE marked board

samuk, 2020-01-18:
I'm looking to get a few nodes placed on buildings. It may be helpful if the boards proposed have CE markings in Europe.

This board is based on ESP32 and has an SX1276 so getting it working may not be too difficult?

https://pycom.io/product/lopy4/

It also appears to have a LoRa interrupt on [Pin23 ](https://docs.pycom.io/gitbook/assets/specsheets/Pycom_002_Specsheets_LoPy_v2.pdf) So the Powersaving referenced [here ](https://github.com/sudomesh/disaster-radio/issues/41#issuecomment-575591440)may be possible

paidforby, 2020-01-18:
Nice find. It's a little expensive, especially when you add all the additional pieces. Looks like a nice board. I ordered one. I will try to get it working once it arrives.

samuk, 2020-01-18:
Great! I spotted the Paxcounter project supports a bunch of boards. Not sure if the pin definitions are useful: https://github.com/cyberman54/ESP32-Paxcounter/tree/master/src/hal

Could these files just be copied over to support the same range of boards?

paidforby, 2020-01-19:
Hmmm, that's cool, they have some sort of HAL (hardware abstraction layer) design. I don't think it would be as simple as copying and pasting these header files, because it's not clear to me how the pin definitions are being used and I do not think they are using the arduino-LoRa library, which has some built-in HAL that might conflict with their code. It does look it will be helpful while investigating other boards. Thanks!

samuk, 2020-01-19:
I thought it might not be as simple as that..

Did you get the GPS addon shield? https://pycom.io/product/pytrack/

samuk, 2020-01-21:
These look interesting too: https://pycom.io/product/pygo1/

I mentioned Disaster radio on the Pycom forum and their [developer seemed interested](https://forum.pycom.io/topic/5605/would-this-mesh-network-run-on-lopy/3)

paidforby, 2020-03-07:
So I have successfully flashed our firmware to a LoPy4. It took a little bit of doing because the LoPy4 does not have an on-board circuit that switches it in to bootloader mode. I found some useful instructions on how to program it [here](https://www.thethingsnetwork.org/labs/story/program-your-lopy-from-the-arduino-ide-using-lmic). Since I didn't have the expansion board and the Pysense board doesn't provide breakouts for the pins, I ended programing it with a a USB-to-UART adapter, I'd recommend following the USB UART Adapter wiring guide shown on [this page](https://docs.pycom.io/gettingstarted/connection/lopy4/).

I still need to figure out the proper pinout for the LoRa transceiver. It should work by default because the pins are defined in the arduino-esp32 library [here](https://github.com/espressif/arduino-esp32/blob/master/variants/lopy4/pins_arduino.h), but it doesn't seem to be working. I'll post to this issue once I sort out pin definitions.

samuk, 2020-05-13:
@paidforby Any progress on this one?

paidforby, 2020-05-13:
Negative, the LoPy4 has been sitting in my closet for months. I think I've been discouraged because of the complicated flashing process, idk why they didn't just put a flashing circuit on their board like everyone else. I'll take another crack at it before June 1.

paidforby, 2020-05-14:
I'm having some trouble getting this working even with a very simple example. I opened an issue here https://github.com/sandeepmistry/arduino-LoRa/issues/364. Maybe someone else has had success, or maybe there's a reason it won't work.

samuk, 2020-05-14:
Annoying. I'll cross post into the Pycom forum if that's OK.. I'll need CE certified hardware for my installs which is why I'm keen on it..

paidforby, 2020-05-14:
I got it working! They were able to solve my issue. I just need to add `SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_CS);` before calling `LoRa.begin()`. I've got it working with our firmware as well, just need to do a little testing. 

Weird that we haven't come across this problem before with other boards, but I guess, they default SPI settings for the LoPy4 are not set to the LORA SPI pins?

I'll push a commint to resolve this issue soon.

paidforby, 2020-05-14:
Interestingly, I had to make `SPI.begin` only be called on LoPy4 as it appears to break the transmitting of packets on the TTGO. 

Anyway, the LoPy4 (an FCC and CE marked board) will be supported as of version 1.0.0-rc.2. If you come across other FCC/CE boards that you would like supported, open a new issue.

closed 2020-05-14 by paidforby

samuk, 2020-05-14:
Great news :) Thanks..

samuk, 2020-05-14:
If you have a multimeter and spare USB cable lying around I'd be interested in the real-world power consumption of this board running Disaster Radio in Bluetooth mode..

paidforby, 2020-05-15:
Can do. I'll take a look at the power consumption when I get a chance, might not happen for a week or two.

h-c-c, 2021-11-19:
Great work folks.  Did you ever use the LoPy4 with the Pycom expansion board?  I can't seem to get the SD card to mount.

samuk, 2021-11-22:
Afraid I haven't. It may be worth digging in https://github.com/cyberman54/ESP32-Paxcounter code, I think Disaster Radio reused their HAL layer & they support the SD.

commit 6f6cc8b8 (2020-05-14): add support for LoPy4, fixes #42