sudomesh/disaster-radio #46: Make the Pins to Connect to LoRa Variables

candrist, 2020-01-24:
In order to support multiple boards and versions, can we make the pins assigned to the LoRa variables that we can assign?

paidforby, 2020-01-25:
Yes, this has been supported since this pull request to LoRaLayer2, https://github.com/sudomesh/LoRaLayer2/pull/1

Just call, `Layer1.setPins(int cs, int reset, int dio);` before `Layer1.initi();`.

We are currently working on expanding support for a number of boards by adapting Paxcounter's HAL (Hardware Abstraction Layer), https://github.com/cyberman54/ESP32-Paxcounter/tree/master/src/hal

If you see your board in that list, then it should be supported soon. Keep an eye out for an 0.1.2 release which may include that expanded support.

candrist, 2020-01-25:
Where would I set MISO, MOSI, and SCK?

paidforby, 2020-01-25:
I have never had to specific MISP, MOSI, and SCK. Perhaps I've always used a board for which the defaults work. Someone else had luck with the syntax like this,

```
#define LORA_SCK SCK
#define LORA_MISO MISO
#define LORA_MOSI MOSI
```
However, they were using the refactored code that is located in PR https://github.com/sudomesh/disaster-radio/pull/44

See https://github.com/sudomesh/disaster-radio/issues/40#issuecomment-578245399 for their complete solution.

morganrallen, 2020-01-25:
I haven't looked at this code in a a while but if the underlying LoRa class object is exposed you should be able to call `setSPI`
See https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-spi-interface

paidforby, 2020-01-25:
Thanks @morganrallen `LoRa.setSPI()` should be accessible. Just make sure you call it before `Layer1.init()` is called (since that calls `LoRa.begin()`) I'm not sure what you will need to pass into `setSPI` for your board, but it should be of the type `SPIClass`.

candrist, 2020-01-29:
Thanks. I was able to get it flashed and working.

closed 2020-01-29 by candrist