sudomesh/disaster-radio #55: Rebroadcast/ Sync messages when network is quiet.

samuk, 2020-02-14:
This would require timestamps on messages 
https://stackoverflow.com/questions/595884/what-is-the-smallest-number-of-bytes-that-can-store-a-timestamp#595930

It also relies on a couple of assumptions:
1) If a node has GPS then it adds timestamps to messages it sends, records timestamps in the history on the sd card.

2) If a message originates from the Bluetooth app a timestamp is added to the messages from the phone/ GPS. Nodes receiving messages via Bluetooth record the timestamp in the sd card history.

Most humans will sleep from ~1am > ~6am local time. So it seems like even in heavily used networks, then during those periods there would be some spare network capacity. If one node in the network sequentially rebroadcasts the last ~300 messages on it's SD card then the whole network should sync.

So something like;

If a node has not received any (R)ebroadcast messages in the last three hours
If there is a SD card present that includes timestamped messages 
If no messages of any kind have been received in the last 60 seconds
Then re-broadcast the last message from the last 300 timestamped messages on the SD, Set byte 15 in the message to (R)etransmit.
Increment by one message 
Repeat

Over a completely quiet 6 hour period approx 360 messages would be synced across the network. If for whatever reason the network has high usage then the rebroadcasting wouldn't happen, but I'd imagine this would be the exception rather than the rule.

Nodes receiving (R)ebroadcast messages would check the timestamp, and discard them if they match an existing timestamp in the history, or insert them in the correct place in the history. 

If the clocks were out you'd get a duplicate message, but that would be better than missing messages?

Nodes with no GPS accessed via Wifi would not get their messages rebroadcast as they would not be timestamped.

The first node added to the network would play the rebroadcast role, as it would be the first to go three hours without receiving an (R) message. If this node goes down then the second node added would pick up the role.

It would be a little bit Scuttlebutt like.

paidforby, 2020-04-15:
Great idea!

>So something like;

> * If a node has not received any (R)ebroadcast messages three hours after it saw an entry in its routing table
> * If there is a SD card present that includes timestamped messages
> * If no messages of any kind have been received in the last 60 seconds
> * Wait a random number of seconds (sum of numbers in the MAC address maybe?)
> * Then re-broadcast the last message received, re-using the original timestamp. Set byte 15 in the message to (R)etransmit.
> * Increment SD log by one message
> * Repeat
>
> One node in any network would have this 'rebroadcast' role at any one time. It would slowly work it's way through the SD history until interrupted by a new message from a human.

Sounds like this could be written as another client or middleware? A little bit like the `History` feature.

> Over a completely quiet 6 hour period approx 360 messages would be synced across the network. If for whatever reason the network has high usage then the rebroadcasting wouldn't happen, but I'd imagine this would be the exception rather than the rule.

The only catch is that routing messages would still be transmitted during this period. Though, maybe that could be accounted for in the rebroadcaster?

> Nodes receiving (R)ebroadcast messages would check the timestamp, and discard them if they match an existing timestamp in the history, or insert them in the correct place in the history.
>
> If the clocks were out you'd get a duplicate message, but that would be better than missing messages?

You could also create a hash of the messages and check if the rebroadcasted message matches the hash of an existing message. Then the timestamp could be used as another layer of verification.

>Nodes with no GPS accessed via Wifi would not get their messages rebroadcast as they would not be timestamped.

This gives me the crazy/stupid idea of implementing some form of meshy-NTP for the nodes that don't have direct access to a real clock (via GPS or BLE). My thought is if a node is receiving timestamped messages it should be able to figure out the approximate time. It may have to account for airtime and duty cycles. Maybe it could even "triangulate" the time by consulting multiple nodes?

> It would be a little bit Scuttlebutt like.

Sure, scuttlebutt is cool.

samuk, 2020-05-13:
Still quite keen on this if it seems feasible. Would help somewhat with lost messages if a given node is busy when it should be receiving a message.

Would the new reactive routing have implications for this idea? Maybe the 'rebroadcast' messages could help with mapping the network somehow?