sudomesh/disaster-radio #58: Tidy up hardware support

samuk, 2020-02-28:
It would be good to tidy up  
https://github.com/sudomesh/disaster-radio/blob/master/firmware/esp32/config.h 

To properly support  at least the boards listed here: 
https://github.com/sudomesh/disaster-radio/wiki/Devices--&-Hardware 

I think the pin definitions needed can be found in these files: 
https://github.com/cyberman54/ESP32-Paxcounter/tree/master/src/hal​​

paulo060507, 2020-02-28:
Good

paidforby, 2020-03-06:
As mentioned in #33 I have tested with the TTGO T-Beam V07 and I would consider that board to now be fully-supported. There's still some work to be done on actual implementation of the GPS client.

I believe that the T-Beam referenced in the `config.h` file is the V10, not V07 because it has pins for the OLED screen which is only on V10. I'm also unsure why the LoRa pin definitions are commented out. We should make a point of differentiating between the V07 and V10.

I have a LoPy4 that I am hoping to test soon. After that, I only know of anecdotal claims for the Heltec V2, Sparkfun 1 channel, and TTGO V1. I'm not sure the status of development on the latest DisasterRadio board.

paulo060507, 2020-03-10:
ok i'm new here but i will be following and i find this question of the oled display very interesting i believe that really only in the last version

marcidy, 2020-06-27:
FYI, #74 included an update to config.h for heltec-v2.

https://github.com/sudomesh/disaster-radio/blob/78dac9bc4edb161b691666e628e756f4a3710925/firmware/config.h#L58-L72

Not clear if that closes this issue.

paidforby, 2020-06-27:
I just made some edits to the wiki that bring it up to date. I think we can close this issue unless @samuk has an objection.

It should also be noted here that most of the pin definitions needed for LoRa and OLED related stuff are defined here https://github.com/espressif/arduino-esp32/tree/master/variants. Which is why the T-Beam pins are commented out and the TTGO-v2 pins are not included in our config.h.

closed 2020-06-27 by paidforby

samuk, 2020-06-30:
Great. This ticket was for someone who volunteered to do some hardware support stuff. They never materialised to do any work, so happy to close now.

marcidy, 2020-06-30:
If you've got more hardware issues, I can take a look.
I can also offer to test on the heltec-v2 boards when mine aren't occupied
with work.
I attempted to launch SSB using Patchwork but the AppImage doesn't work for
me and promptly decided to do something else than deal with it.  If that's
the best place to keep abreast of discussion I'll figure it out.  If you
post more tickets tagged with help-wanted and/or hardware I'll look at
those directly.

paidforby, 2020-07-01:
Cool, that's good to know. This is probably the best place to stay in the loop on firmware related stuff. I'm not that active on SSB, though I've also been meaning to get back into it. (I always run into problems like that with SSB too and get discouraged quickly).