sudomesh/disaster-radio #59: Serve user defined files from SD card

samuk, 2020-02-28:
It would be interesting to serve files such as the Android app https://play.google.com/store/apps/details?id=tk.giesecke.disaster_radio

From the SD when in Wifi mode. Is it feasible to just serve up whatever arbitrary files the user adds to the SD?

paidforby, 2020-03-05:
This is as simple as adding the file to the `/web/static` directory on the SD card. Then, just request the page http://192.168.4.1/filename and you should automatically download the file. Obviously, we may want to eventually build some sort of file browser/downloader into the web app to make this feature more accessible, but as a proof of concept this would get you started.

samuk, 2020-03-05:
Thanks I'll try copying the webapp to SD.

Would it be possible to do it even when the webapp is hosted on the spiffs filesystem?

paidforby, 2020-03-05:
Yes, the same concept applies, obviously the size of the file is a greater concern if you are using SPIFFS. Just copy the desired fiie into `web/static/` after building the webapp, but prior to running `pio run -t uploadfs` and it will be written to the SPIFFS along with the webapp.

closed 2020-04-24 by samuk