sudomesh/disaster-radio #61: I have been installed in a Heltec V2 and now?

AnakinSpain, 2020-03-21:
I am a IOS user, but after installed in a Heltec V2 in 192.168.4.1 blank page.

And now what I  can do?

paidforby, 2020-03-21:
You mostly likely need to flash the SPIFFS also. First you need to build the app using npm, you can find instruction for that [here](https://github.com/sudomesh/disaster-radio/tree/master/web#setup). Then you need to flash the built web app to your board with `pio run -e heltech-v2 -t uploadfs`. There are instructions for flashing the SPIFFS [here](https://github.com/sudomesh/disaster-radio#building-web-app). 

Finally, if you have trouble building the web app with npm, you can download the pre-built web app as part of the [latest release](https://github.com/sudomesh/disaster-radio/releases) and then copy the files `web/static` from the zip into the `web/static` directory in the cloned repo.

samuk, 2020-05-13:
@AnakinSpain did this work for you? Closing the ticket but please reopen if you're still having problems.

closed 2020-05-13 by samuk