sudomesh/disaster-radio #62: Maximum hops between nodes

D4rk4, 2020-03-22:
I plan setup network based on TTGOv2-433, and I got question. How much maximum hops can be between nodes in one network for sucessful transmit messages?
In routing page in disaster-radio wiki it's not clear.

paidforby, 2020-03-26:
This has not been thoroughly tested. The limit on the number of the routes that can be transmitted in a single packet is currently 30, however an upcoming change should increase that number to 40. So those are the theoretical maximum number of nodes in a network right now. It's possible that we could make improvements to the routing protocol or routing table management that could increase the theoretical size, see more about alternative routing options in #57. 

While there is a limited routing table size, there is theoretically no limit on the number of hops since we use a naive routing protocol (i.e. nodes only care about the next hop, not the entire route, to get to a destination). For example, you could have 30 nodes lined up and the protocol should eventually discover each end of the line and be able to route over all thirty hops. I encourage you to play around with https://github.com/sudomesh/disaster-radio-simulator to test the limits of the protocol (note: the simulator has not been synced with the latest changes to LoRaLayer2, so it may have some bugs).

samuk, 2020-04-29:
@D4rk4 closing as I think your question is answered? Please re-open if not

closed 2020-04-29 by samuk