sudomesh/disaster-radio #64: Small inconsistency in the website

gioblu, 2020-03-31:
Here: https://disaster.radio/learn/routing/

>(i.e. the number of packets transmitted and the amount of time spent "on-air")

The overhead is the amount of data added to information required to carry out the exchange.

paidforby, 2020-03-31:
You may be correct when talking specifically about network overhead. I suppose I am referring to the general overhead that should be considered in LoRa networks. For what it's worth, the wikipedia definition of [overhead](https://en.wikipedia.org/wiki/Overhead_%28computing%29) is "any combination of excess or indirect computation time, memory, bandwidth, or other resources that are required to perform a specific task." I would consider airtime to be a resource required to transmit over LoRa. The more packets you send, the more time you spend on air, and the less of that resource that is available for others to transmit.

If you would like to suggest a revision to make this more clear and accurate, I'm open to ideas. Otherwise, I will think of a way to restate it.

gioblu, 2020-03-31:
Absolutely, I may have been misled by the sentence's subject
> This protocol is designed to be as minimal as possible in order to reduce network overhead 

What I mean is, If you have tried to keep the protocol as minimal as possible, you have taken care to keep the protocol's overhead low. Although I agree with you, this is not a major issue :)

Compliments for the great work done on this project.
Keep up the good work 👍

closed 2020-03-31 by gioblu