sudomesh/disaster-radio #67: Better documentation for solar node and hardware

jooray, 2020-04-10:
Being a software guy that does not understand hardware very well yet, I would like a written set of recommendations for building a solar node (I believe dev board is covered by https://www.aliexpress.com/item/4000396836096.html ).

I would like to suggest an enhancement to https://github.com/sudomesh/disaster-radio/wiki/Devices--&-Hardware section Complete Solar nodes.

I would like a description of each component and how it relates to other components. What parameters to look for and what are the alternatives.

The wishlist from Aliexpress is nice, but it misses a type and quantities.

For example:
 - Waterproof box: https://www.aliexpress.com/item/4000429348282.html?spm=0.8937489.0.0.47902e0e5UYCHn - no longer available. I found https://www.aliexpress.com/item/4000442364567.html as an alternative. Will it be large enough? 4-8mm or 9-12mm? (I guess 4-8mm should be enough given the size of the cable, but what about the antenna?)

 - Cable: https://www.aliexpress.com/item/33053500652.html?spm=0.8937489.0.0.67042e0esfbOd8 - 2 cores? 3 cores? 4 cores?

 - Solar panel: https://www.aliexpress.com/item/4000031921317.html?spm=0.8937489.0.0.19bd2e0emPN0dl - the description mentions 20W, but the store does not ship this to my country. What parameters and  connectors are needed? For example, would this work as an alternative? https://www.aliexpress.com/item/32847502994.html Parameters seem to be similar, but I don't know about the connectors (I am fine with soldering, but I don't know what are the standard connectors or if I should solder).

- Batteries - https://www.aliexpress.com/item/32693860069.html?spm=0.8937489.0.0.67042e0euw6kKt I need four pieces, right?

- Controller - https://www.tindie.com/products/xorbit/lifepo4weredsolar1/ - default charge - what should I use? Thermistor - yes or no?

What I suggest is describing each component, how many pieces and people could submit alternatives (stores) available locally.

People could also send component kits, I would happily pay for that.

(I am a part of local city community that would like to deploy this city-wide and play with the project. There are a few 3D printing gurus, more than a few software developers, but we would really like some guidance in the hardware part).

samuk, 2020-04-10:
Hi Juraj that page is largely my fault

> I would like a description of each component and how it relates to other components. What
> parameters to look for and what are the alternatives.

Yeah fair enough, i'll try an improve that page. There's a bit more here:
https://hackaday.io/project/170069-disaster-radio-lora-mesh

> * Waterproof box:

I've just re-added a box that's in stock. The 8mm gland should work.

> * Cable: https://www.aliexpress.com/item/33053500652.html?spm=0.8937489.0.0.67042e0esfbOd8 - 2
> cores? 

2 core, you just get a positive and a negative from a solar panel.
 
> * Solar panel: https://www.aliexpress.com/item/4000031921317.html?spm=0.8937489.0.0.19bd2e0emPN0dl
> - the description mentions 20W, but the store does not ship this to my country. What parameters and
> connectors are needed? For example, would this work as an alternative?

You just need two bare ends of wire, to attach to the screw terminals on the solar controller. (Assuming you ask for the screw terminals pre-soldered by Xorbit, he charges a couple of dollars for this)

> https://www.aliexpress.com/item/32847502994.html Parameters seem to be similar, but I don't know
> about the connectors (I am fine with soldering, but I don't know what are the standard connectors
> or if I should solder).

That will work for a bit, the plastic covering degrades fairly rapidly on those flexible panels which is why I specified the glass panel which _should_ last a bit longer before it fails.

> * Batteries - https://www.aliexpress.com/item/32693860069.html?spm=0.8937489.0.0.67042e0euw6kKt I
> need four pieces, right?

Probably. If you live on the equator one or two may work. In the Arctic you may need 10+. 

I've specced this for 1hour of sunlight/ day which is what we get in January in midwinter in the UK.

> * Controller - https://www.tindie.com/products/xorbit/lifepo4weredsolar1 - default charge - what
> should I use? 

It depends what panel you add, if you use the 20w then the 2.1A will be fine. If you use a 10w panel you should get away with the 0.5A

>Thermistor - yes or no?

Yes, get the one he recommends

> What I suggest is describing each component, how many pieces and people could submit alternatives
> (stores) available locally.

Yeah fair enough. It needs more work and testing.

Thanks

Sam

samuk, 2020-04-10:
I've made some changes to this page https://github.com/sudomesh/disaster-radio/wiki/Devices--&-Hardware 

And the Aliexpress wishlist, which I think addresses these concerns, please re-open the ticket if not.

closed 2020-04-10 by samuk

jooray, 2020-04-12:
Thank you, this description is amazing.