sudomesh/disaster-radio #70: LoStik + Linux application?

paidforby, 2020-04-26:
While the updating the simulator i got the idea that it might be worth it to write a real linux application that could interface with a real disaster radio network, not just a simulated network.

As it stands, I've ported almost the entire disaster radio firmware to generic C++. You can see that the [simulator firmware](https://github.com/sudomesh/disaster-radio-simulator/tree/master/routers),  is now very similar to the [esp32 firmware](https://github.com/sudomesh/disaster-radio/tree/master/firmware/esp32). They are so similar that I may eventually merge the two.

So the main challenge would be to find a LoRa transceiver that has good Linux support and to write another Layer1 interface for it. 

I came across the LoStik, https://www.crowdsupply.com/ronoth/lostik and was curious if anyone has experience with it or has other recommendations for a LoRa-to-USB device, such as this one, https://www.tindie.com/products/allankliu/low-cost-lora-usb-dongle-with-stm32/.

samuk, 2020-04-27:
I'm sure @tlrobinson got a Lostik working? I can't find the ticket now though?

Bovril1a, 2020-04-29:
Re the Lostik it "relatively" easily allows for setting up TCP/IP connectivity via PPP

https://github.com/jgoerzen/lorapipe/blob/master/doc/lorapipe.1.md

I managed after some support from jgoerzen to set two Raspberry Pi's (3B+ and 4) up and ran (dial up speed) SSH and even opened an *incredibly* slow VNC session between them.. 8-) Even robust enough to run a short WireGuard VPN session between the two.

If we can set up the same sort of USB, plug and play (ish) functionality in the LORA32's and T-Beams, could add substantially to ease of porting messaging and other functionality

samuk, 2020-05-06:
@paidforby Here's the config @tlrobinson used: https://github.com/sudomesh/disaster-radio/issues/50

paidforby, 2020-05-06:
Good catch. Some how that issue escaped my awareness. I should be able to adapt the example in that issue, or at least use its documentation, to get the LoStik working with the simulator code as an initial step toward a Linux application. Thanks.