sudomesh/disaster-radio #71: Linux Client For BLE Connection To Node

pqhf5kd, 2020-05-03:
Is anyone planning a Linux client to talk to a node over BLE like the [Android app](https://github.com/beegee-tokyo/disaster-radio-android)?

paidforby, 2020-05-03:
@pqhf5kd That's a great idea! I, personally, do not have plans for a writing a Linux BLE application, but I encourage someone to give it a shot. I'm not sure what all would be involved, but I imagine it could connect to the same client code as the Android app, i.e. [BleUartClient.cpp](https://github.com/sudomesh/disaster-radio/blob/master/firmware/esp32/client/BleUartClient.cpp). That was written by @beegee-tokyo, who may have some insights about what would be needed in making a Linux application work with it.

beegee-tokyo, 2020-05-03:
A Linux BLE app for DisasterRadio should be not to difficult. The communication is using standard BLE UART characteristic, so all needed there is to parse the messages. The protocol is similar to the one of the Web Socket interface.
More challenging might be the UI. 

I personally have zero experience with writing Linux apps.

paidforby, 2020-05-06:
I may eventually look in to building a Linux app of some kind if I start working on supporting the LoStik, as I mentioned in #70. I _feel_ that node.js/Electron could be a good route to pursue for a Linux app, but I don't have much experience with building Linux applications either.