sudomesh/disaster-radio #72: Replace Preferences.h with more genral settings management

paidforby, 2020-05-15:
I'm considering replacing the current method of storing the settings. Currently only username and interface mode are stored using the arduino-esp32 specific Preferences.h.

I believe it was @beegee-tokyo who introduced the Preferences.h. I was wondering if you had a good reason to use Preferences over writing to a normal txt, json, or header file? I suppose it uses non-volatile storage, which is safer, but has the drawback of not being portable to other devices.

I'm actively considering two options,

A) port Preferences.h to standard cpp and have it write to a text file instead of esp32 non-volatile storage. Or provide an option for writing to one or the other.

B) come up with a different method for fetching and storing settings, possibly using the existing Console line processing functionality. For example, on-boot, at the very beginning of setup, a node feeds a series of "commands" from a text file into a Console middleware object. Maybe this could involve creating a "Configuration client" that behaves very similarly to the serial or tcp clients except that is reads from and and writes to a text file.

Curious if anyone has strong opinions on this before I spend time on one of these options.

beegee-tokyo, 2020-05-18:
@paidforby    
The ESP32 has neither NVM nor EEPROM. Preferences lib is using a dedicated flash partition. 
I chose it to avoid that saved settings are erased from SPIFFS or similar file system simulations by accident (e.g. updating the Webserver files).
But feel free to replace it with a more Arduino-like solution.

paidforby, 2020-05-19:
Got it. That makes sense. That's why you added the custompart.csv, right? Thanks for that explanation, I'll think about it as I consider some options. I actually want a less Arduino-like solution and a more general C/C++ solution, with the idea that it could also work with the [simulator](https://github.com/sudomesh/disaster-radio-simulator) or a Linux application as mentioned in #70 and #71

paidforby, 2020-05-20:
I've temporarily side-stepped my problem by using a few well-placed `#ifdef`s. After further thought, I do like the Preferences method, but I would like to find something similar for use on Linux and rather than replacing Preferences, just provide an alternative.