sudomesh/disaster-radio #78: Power and antennas

jooray, 2020-08-04:
Hi,

we are trying to bootstrap a disaster radio network in Bratislava, Slovakia, but it seems that the dev board has very low transmission power, it broadcasts only maybe up to a 100 meters.

Is there a setting to set transmission power and stay within regulatory limits? Or do we need to build somehow firmware manually?

Or is it something that is just resolved by a better antenna? Are there any particular you recommend in an urban environment?

Just to be clear, what we don't know is if we need to set transmission power in software or we just need to upgrade an antenna.

If I get an answer, I will update docs for new users.

Thanks,
Juraj.

paidforby, 2020-08-07:
Are you using the TTGOv2 board? If you are using the very small antenna that comes with that board (I'm not sure the exact gain of that antenna but it isn't good), I'd definitely recommend a better antenna, I've seen packets transmit a few miles with decent antenna and a good high locations. The transmit power defaults to 17dB, this can be increased to 20dB. I'm not sure of the regulations surrounding txPower (especially in Slovakia), but you can change the txPower by calling `Layer1->setTxPower(int txPower);`. before `Layer1->init()` in `main.ino` and then rebuild the firmware. There is currently no way to change the txPower without rebuilding the firmware and re-flashing the board (though the option to set txPower through the console should be added).

jooray, 2020-08-14:
Yes, TTGOv2. Also raising spreading factor could be great, but for that we all have to rebuild the firmware. Do you think raising it to 11 by default for the city-wide network is a bad idea? Not everyone can put it high enough.

I use a different antenna than what I got, but we could find better.

Changing all the parameters (transmit power, spreading factor) via telnet console should be a good way for experiments, although the network has to agree on a single SF in the end.

samuk, 2020-08-14:
Nagoya antennas seem very well regarded.

If you could find a supplier for these in Europe I'd be interested to know about them: https://www.nagoya.com.tw/en/2-2277-66934/product/868MHz-NA-868-id401348.html

paidforby, 2020-08-18:
@jooray just a heads up, recent commits (including the one mentioned above) adds the configuration option of setting txPower through the console using the command `/set txpower VAL`. This setting is saved in separate partition of flash memory and is recalled on subsequent boots (even after reflashing).

Now that I've figured out the system I want to use (a large thanks to @beegee-tokyo who initially implemented the `Preferences` system), it should be simple to add any number of configurable features to the `/set` command so that is cool. Probably next I'll add spreading factor and lora frequency.

If you feel your inital questions have been answered, feel free to close this issue. Thanks!

Note: I will probably make a binary build of 1.0.0-rc.2 after I get a few more of the configuration options implemented, but still it might not happen for a few weeks. So I recommend building from master instead of waiting on me to write up a proper release.

jooray, 2020-08-19:
Amazing, thank you very much!

closed 2020-08-19 by jooray