sudomesh/disaster-radio #79: build firmware for ttgo-lora32-v2 failed.

wxie2017, 2020-08-06:
wxie@guix ~/work/disaster-radio/firmware$ ~/.local/bin/pio run --verbose
Processing ttgo-lora32-v2 (board: ttgo-lora32-v1; build_flags: -DTTGO_LORA_V2, -DARDUINO_LORA, -DHAS_OLED, -I./src; platform: https://github.com/platformio/platform-espressif32.git; framework: arduino; upload_port: /dev/ttyUSB0; monitor_port: /dev/ttyUSB0; monitor_speed: 115200; board_build.partitions: custompart.csv; lib_deps: AsyncTCP, ESP Async WebServer@1.2.3, LoRa@0.7.2, https://github.com/jgromes/RadioLib#3682c6c9215891e3afb7672f1235fde1c3bd75fd, https://github.com/sudomesh/LoRaLayer2#1e2204ebaa7f81b8d06853fedb92ab0ddc9b9414, https://github.com/paidforby/AsyncSDServer#13375c6be978cb34180378ecf4042a3a4a1f5eab, ESP8266 and ESP32 OLED driver for SSD1306 displays, TinyGPSPlus@1.0.2, ArduinoJson)
--------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-lora32-v1.html
PLATFORM: Espressif 32 1.12.4 #08ef4bf (git+https://github.com/platformio/platform-espressif32.git) > TTGO LoRa32-OLED V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
xtensa-esp32-elf-g++ -o "/home/wxie/work/disaster-radio/firmware/main.ino.cpp" -x c++ -fpreprocessed -dD -E "/tmp/tmpr09_spwu"
sh: /home/wxie/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: Datei oder Verzeichnis nicht gefunden
*** Error 127
AssertionError: :
  File "/home/wxie/.local/lib/python3.8/site-packages/platformio/builder/main.py", line 168:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/wxie/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/wxie/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/wxie/.platformio/platforms/espressif32/builder/main.py", line 223:
    target_elf = env.BuildProgram()
  File "/home/wxie/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/home/wxie/.local/lib/python3.8/site-packages/platformio/builder/tools/platformio.py", line 61:
    env.ProcessProgramDeps()
  File "/home/wxie/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/home/wxie/.local/lib/python3.8/site-packages/platformio/builder/tools/platformio.py", line 118:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/home/wxie/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/home/wxie/.local/lib/python3.8/site-packages/platformio/builder/tools/platformio.py", line 327:
    env.ConvertInoToCpp()
  File "/home/wxie/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/home/wxie/.local/lib/python3.8/site-packages/platformio/builder/tools/piomisc.py", line 240:
    out_file = c.convert(ino_nodes)
  File "/home/wxie/.local/lib/python3.8/site-packages/platformio/builder/tools/piomisc.py", line 94:
    return self.process(contents)
  File "/home/wxie/.local/lib/python3.8/site-packages/platformio/builder/tools/piomisc.py", line 115:
    assert self._gcc_preprocess(contents, out_file)
======================================================== [FAILED] Took 0.66 seconds ========================================================

Environment        Status    Duration
-----------------  --------  ------------
ttgo-lora32-v1     IGNORED
ttgo-lora32-v2     FAILED    00:00:00.661
ttgo-lora32-tbeam  IGNORED
heltec-v2          IGNORED
sparkfun-lora      IGNORED
lopy4              IGNORED
================================================== 1 failed, 0 succeeded in 00:00:00.661 ==================================================

What's the problem?

paidforby, 2020-08-07:
I'm now sure what the error is here, looks like it's not finding the file `/home/wxie/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++`? This must be a problem with your build environment because my tests in travisCI pass, https://travis-ci.org/github/sudomesh/disaster-radio. Maybe make sure platformio is up to date (or maybe downgrade it)? I am using `PlatformIO, version 4.3.5a1`, you can find this by running `pio --version`.

Maybe try using the stable version of espressif32 by changing line 23 in platformio.ini to,
```
platform = espressif32
```

If this doesn't work, maybe figure out how to install the xtensa toolchain manually. This should be done automatically by platformio, but it seems like yours is not being installed properly (or it is installed in the wrong location?),

wxie2017, 2020-08-09:
Thanks.

1. installation
wxie@guix ~/work/disaster-radio/firmware$ sudo pip install -U platformio
Passwort: 
Defaulting to user installation because normal site-packages is not writeable
Collecting platformio
  Downloading platformio-4.3.4.tar.gz (190 kB)
     |████████████████████████████████| 190 kB 5.9 kB/s 
Collecting bottle<0.13
  Downloading bottle-0.12.18-py3-none-any.whl (89 kB)
     |████████████████████████████████| 89 kB 22 kB/s 
Collecting click<8,>=5
  Downloading click-7.1.2-py2.py3-none-any.whl (82 kB)
     |████████████████████████████████| 82 kB 12 kB/s 
Collecting colorama
  Downloading colorama-0.4.3-py2.py3-none-any.whl (15 kB)
Collecting pyserial!=3.3,<4,>=3
  Downloading pyserial-3.4-py2.py3-none-any.whl (193 kB)
     |████████████████████████████████| 193 kB 12 kB/s 
Collecting requests<3,>=2.4.0
  Downloading requests-2.24.0-py2.py3-none-any.whl (61 kB)
     |████████████████████████████████| 61 kB 14 kB/s 
Collecting semantic_version<3,>=2.8.1
  Downloading semantic_version-2.8.5-py2.py3-none-any.whl (15 kB)
Collecting tabulate<1,>=0.8.3
  Downloading tabulate-0.8.7-py3-none-any.whl (24 kB)
Collecting pyelftools<1,>=0.25
  Downloading pyelftools-0.26-py2.py3-none-any.whl (136 kB)
     |████████████████████████████████| 136 kB 17 kB/s 
Collecting marshmallow>=2
  Downloading marshmallow-3.7.1-py2.py3-none-any.whl (45 kB)
     |████████████████████████████████| 45 kB 12 kB/s 
Collecting urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1
  Downloading urllib3-1.25.10-py2.py3-none-any.whl (127 kB)
     |████████████████████████████████| 127 kB 17 kB/s 
Collecting idna<3,>=2.5
  Downloading idna-2.10-py2.py3-none-any.whl (58 kB)
     |████████████████████████████████| 58 kB 10 kB/s 
Collecting certifi>=2017.4.17
  Downloading certifi-2020.6.20-py2.py3-none-any.whl (156 kB)
     |████████████████████████████████| 156 kB 11 kB/s 
Collecting chardet<4,>=3.0.2
  Downloading chardet-3.0.4-py2.py3-none-any.whl (133 kB)
     |████████████████████████████████| 133 kB 9.9 kB/s 
Installing collected packages: bottle, click, colorama, pyserial, urllib3, idna, certifi, chardet, requests, semantic-version, tabulate, pyelftools, marshmallow, platformio
  WARNING: The script chardetect is installed in '/root/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script tabulate is installed in '/root/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
    Running setup.py install for platformio ... done
Successfully installed bottle-0.12.18 certifi-2020.6.20 chardet-3.0.4 click-7.1.2 colorama-0.4.3 idna-2.10 marshmallow-3.7.1 platformio-4.3.4 pyelftools-0.26 pyserial-3.4 requests-2.24.0 semantic-version-2.8.5 tabulate-0.8.7 urllib3-1.25.10

2. version
wxie@guix ~/work/disaster-radio/firmware$ ~/.local/bin/pio --version
PlatformIO, version 4.3.4

3. tools
wxie@guix ~/work/disaster-radio/firmware$ ls -al ~/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++
-rwxr-xr-x 1 wxie users 806184  9. Jan 2018  /home/wxie/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++
wxie@guix ~/work/disaster-radio/firmware$ ~/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++ --version
bash: /home/wxie/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: Datei oder Verzeichnis nicht gefunden
wxie@guix ~/work/disaster-radio/firmware$ ls -al ~/.platformio/packages/toolchain-xtensa32/bin/
insgesamt 21188
drwxr-xr-x 2 wxie users    4096  6. Aug 16:59 ./
drwx------ 8 wxie users    4096  6. Aug 16:59 ../
-rwxr-xr-x 1 wxie users  860872  9. Jan 2018  xtensa-esp32-elf-addr2line
-rwxr-xr-x 1 wxie users  889104  9. Jan 2018  xtensa-esp32-elf-ar
-rwxr-xr-x 1 wxie users 1197000  9. Jan 2018  xtensa-esp32-elf-as
lrwxrwxrwx 1 wxie users      20  6. Aug 16:59 xtensa-esp32-elf-c++ -> xtensa-esp32-elf-g++
[... 61 lines cut ...]
  File "/home/wxie/.local/lib/python3.8/site-packages/platformio/builder/tools/platformio.py", line 327:
    env.ConvertInoToCpp()
  File "/home/wxie/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/home/wxie/.local/lib/python3.8/site-packages/platformio/builder/tools/piomisc.py", line 240:
    out_file = c.convert(ino_nodes)
  File "/home/wxie/.local/lib/python3.8/site-packages/platformio/builder/tools/piomisc.py", line 94:
    return self.process(contents)
  File "/home/wxie/.local/lib/python3.8/site-packages/platformio/builder/tools/piomisc.py", line 115:
    assert self._gcc_preprocess(contents, out_file)
======================================================== [FAILED] Took 0.69 seconds ========================================================

Environment        Status    Duration
-----------------  --------  ------------
ttgo-lora32-v1     IGNORED
ttgo-lora32-v2     FAILED    00:00:00.688
ttgo-lora32-tbeam  IGNORED
heltec-v2          IGNORED
sparkfun-lora      IGNORED
lopy4              IGNORED
================================================== 1 failed, 0 succeeded in 00:00:00.688 ==================================================

I use guix 1.1.0

How to solve it?

rawesomeawesome, 2020-08-14:
`C:\Users\rossm\Desktop\disaster-radio-master\firmware>pio run --verbose
Processing ttgo-lora32-v2 (board: ttgo-lora32-v1; build_flags: -DTTGO_LORA_V2, -DARDUINO_LORA, -DHAS_OLED, -I./src; platform: espressif32; framework: arduino; upload_port: /dev/ttyUSB0; monitor_port: /dev/ttyUSB0; monitor_speed: 115200; board_build.partitions: custompart.csv; lib_deps: AsyncTCP, ESP Async WebServer@1.2.3, LoRa@0.7.2, https://github.com/jgromes/RadioLib#3682c6c9215891e3afb7672f1235fde1c3bd75fd, https://github.com/sudomesh/LoRaLayer2#1e2204ebaa7f81b8d06853fedb92ab0ddc9b9414, https://github.com/paidforby/AsyncSDServer#13375c6be978cb34180378ecf4042a3a4a1f5eab, ESP8266 and ESP32 OLED driver for SSD1306 displays, TinyGPSPlus, ArduinoJson)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-lora32-v1.html
PLATFORM: Espressif 32 1.12.4 #08ef4bf (git+https://github.com/platformio/platform-espressif32.git) > TTGO LoRa32-OLED V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
xtensa-esp32-elf-g++ -o "C:\Users\rossm\Desktop\disaster-radio-master\firmware\main.ino.cpp" -x c++ -fpreprocessed -dD -E "C:\Users\rossm\AppData\Local\Temp\tmpkgnrm5e6"
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Platform incompatible library C:\Users\rossm\Desktop\disaster-radio-master\firmware\.pio\libdeps\ttgo-lora32-v2\ESPAsyncTCP
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
Looking for ESP8266 and ESP32 OLED driver for SSD1306 displays library in registry
Found: https://platformio.org/lib/show/2978/ESP8266 and ESP32 OLED driver for SSD1306 displays
LibraryManager: Installing id=2978
Using cache: C:\Users\rossm\.platformio\.cache\bc\aa3b625ae9057978e0e77bd37df945bc
ESP8266_SSD1306 @ 4.1.0 has been successfully installed!
Looking for TinyGPSPlus library in registry
Conflict: More than one library has been found by request {"name": "TinyGPSPlus", "requirements": null}:
TinyGPSPlus
===========
#ID: 1655
A new, customizable Arduino NMEA parsing library

Keywords: gps, nmea
Compatible frameworks: Arduino
Compatible platforms: Infineon XMC, Kendryte K210, GigaDevice GD32V, ASR Microelectronics ASR605x, Atmel AVR, Atmel SAM, Espressif 8266, Intel ARC32, Microchip PIC32, Nordic nRF51, ST STM32, Teensy, TI MSP430, TI TIVA, Espressif 32, Nordic nRF52, ST STM8, Atmel megaAVR
Authors: Mikal Hart

TinyGPSPlus
=========UnicodeEncodeError: 'charmap' codec can't encode characters in position 9-12: character maps to <undefined>:
==
  File "C:\Users\rossm\AppData\Local\Programs\Python\Python38-32\Lib\site-packages\platformio\builder\main.py", line 168:
#ID: 8940    env.SConscript("$BUILD_SCRIPT")

A port of Mikal Hart's TinyGPS++ version 1.0.2 to mbed.
  File "C:\Users\rossm\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:

    return _SConscript(self.fs, *files, **subst_kw)
Compatible frameworks: Mbed
Compatible platforms: Atmel SAM, Freescale Kinetis, Nordic nRF51, NXP LPC, Silicon Labs EFM32, ST STM32, Teensy, Nordic nRF52, Maxim 32, WIZNet W7500, RISC-V GAP, NXP i.MX RT
  File "C:\Users\rossm\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
Authors: Shigeru Fujiwara
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)

  File "C:\Users\rossm\.platformio\platforms\espressif32\builder\main.py", line 223:
TinyGPSPlus
    target_elf = env.BuildProgram()
====  File "C:\Users\rossm\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
===    return self.method(*nargs, **kwargs)
===  File "C:\Users\rossm\AppData\Local\Programs\Python\Python38-32\Lib\site-packages\platformio\builder\tools\platformio.py", line 62:
=
    env.ProcessProjectDeps()
#ID: 4917
  File "C:\Users\rossm\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
mbed port of the TinyGPSPlus library for Arduino, by Mikal Hart. The port is heavily inspired by Sergey Drobyshevskiy (dROb)'s port of the TinyGPS library.
    return self.method(*nargs, **kwargs)

Keywords: gps, mkt3339, tinygps
  File "C:\Users\rossm\AppData\Local\Programs\Python\Python38-32\Lib\site-packages\platformio\builder\tools\platformio.py", line 131:
Compatible frameworks: Mbed
    project_lib_builder = env.ConfigureProjectLibBuilder()
Compatible platforms: Atmel SAM, Freescale Kinetis, Nordic nRF51, NXP LPC, Silicon Labs EFM32, ST STM32, Teensy, Nordic nRF52, Maxim 32, WIZNet W7500, RISC-V GAP, NXP i.MX RT
  File "C:\Users\rossm\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\rossm\AppData\Local\Programs\Python\Python38-32\Lib\site-packages\platformio\builder\tools\piolib.py", line 1036:
[... 7 lines cut ...]
  File "c:\users\rossm\appdata\local\programs\python\python38-32\lib\site-packages\platformio\commands\lib.py", line 622:
    click.echo(
[... cut ...]
======================================================================= [FAILED] Took 1.27 seconds =======================================================================

Environment        Status    Duration
-----------------  --------  ------------
ttgo-lora32-v1     IGNORED
ttgo-lora32-v2     FAILED    00:00:01.274
ttgo-lora32-tbeam  IGNORED
heltec-v2          IGNORED
sparkfun-lora      IGNORED
lopy4              IGNORED
================================================================== 1 failed, 0 succeeded in 00:00:01.274 ==================================================================`

rawesomeawesome, 2020-08-14:
i'm not sure what purpose tinygps serves in this build, but i found that by commenting out all references to tinygpsplus i was able to compile just fine

paidforby, 2020-08-16:
Oh I see your problem now. Are you changing the `platformio.ini` to use `board = ttgo-lora32-v2`? That won't work. It's confusing and weird, but I've always used `board = ttgo-lora32-v1` for compiling and flashing to my LILY TTGO v2 board. I'm not sure why I do this, but I vaguely recall having issues when using v2. Notice in my [platformio.ini](https://github.com/sudomesh/disaster-radio/blob/master/firmware/platformio.ini#L51) even for the v2 environment I still use the v1 board definition. 

Does the board you are using only work with the v2 board definition? Have either of you tried with using v1?

Interesting @rawesomeawesome your discovery about TinyGPS, that is not used for anything in the on the TTGO v2  dev board because it doesn't have a GPS. Maybe we could put a `HAS_GPS` build flag to control whether or not gps related code is compiled. I'd be interested to see what all you had to comment out.

paidforby, 2020-08-16:
@wxie2017 you are on older version of platformio (PlatformIO, version 4.3.4). Would you mind updating to the dev branch by running the following,
```
pio upgrade --dev
pio update
```
And see if that makes a difference. Everything else about your environment looks that same. You could also try @rawesomeawesome suggestion of removing all references to TinyGPS.

paidforby, 2020-08-16:
@rawesomeawesome you are not pulling tagged version of TinyGPS, which means you are mostly likely using the latest commit, that is probably why it isn't working. Try pulling the version `1.0.2` (https://github.com/mikalhart/TinyGPSPlus/releases/tag/v1.0.2) by putting `@1.0.2` after the library, like in my [platformio.ini](https://github.com/sudomesh/disaster-radio/blob/master/firmware/platformio.ini#L40).

paidforby, 2020-08-16:
@wxie2017 I downgraded my platformio to 4.3.4 and had no problems compiling. Is there anyway you can try compiling in a clean environment (like a VM)? I've never used guix, I'm mostly on Ubuntu 18.04.

wxie2017, 2020-08-18:
> @wxie2017 you are on older version of platformio (PlatformIO, version 4.3.4). Would you mind updating to the dev branch by running the following,
>
> ```
> pio upgrade --dev
> pio update
> ```
>
> And see if [...]

The upgrade failed:

wxie@guix ~/work/disaster-radio/firmware$ ~/.local/bin/pio upgrade --dev
Please wait while upgrading PlatformIO ...
Error: ['/gnu/store/11l2qmzfgsp7k345mv6x1vn64q8330kw-python-wrapper-3.8.2/bin/python', '-m', 'pip', '--no-cache-dir', 'install', '--upgrade', '/home/wxie/.platformio/.cache/piocoredevelop.zip']

Traceback (most recent call last):
  File "/gnu/store/09a5iq080g9b641jyl363dr5jkkvnhcn-python-3.8.2/lib/python3.8/runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/gnu/store/09a5iq080g9b641jyl363dr5jkkvnhcn-python-3.8.2/lib/python3.8/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/wxie/.guix-profile/lib/python3.8/site-packages/pip/__main__.py", line 16, in <module>
    from pip._internal import main as _main  # isort:skip # noqa
  File "/home/wxie/.guix-profile/lib/python3.8/site-packages/pip/_internal/__init__.py", line 40, in <module>
    from pip._internal.cli.autocompletion import autocomplete
  File "/home/wxie/.guix-profile/lib/python3.8/site-packages/pip/_internal/cli/autocompletion.py", line 8, in <module>
    from pip._internal.cli.main_parser import create_main_parser
  File "/home/wxie/.guix-profile/lib/python3.8/site-packages/pip/_internal/cli/main_parser.py", line 11, in <module>
    from pip._internal.commands import (
  File "/home/wxie/.guix-profile/lib/python3.8/site-packages/pip/_internal/commands/__init__.py", line 6, in <module>
    from pip._internal.commands.completion import CompletionCommand
  File "/home/wxie/.guix-profile/lib/python3.8/site-packages/pip/_internal/commands/completion.py", line 6, in <module>
    from pip._internal.cli.base_command import Command
  File "/home/wxie/.guix-profile/lib/python3.8/site-packages/pip/_internal/cli/base_command.py", line 26, in <module>
    from pip._internal.index import PackageFinder
ImportError: cannot import name 'PackageFinder' from 'pip._internal.index' (/home/wxie/.guix-profile/lib/python3.8/site-packages/pip/_internal/index/__init__.py)

* Upgrade using `pip install -U platformio`
* Try different installation/upgrading steps:
  https://docs.platformio.org/page/installation.html

wxie2017, 2020-08-18:
> @wxie2017 I downgraded my platformio to 4.3.4 and had no problems compiling. Is there anyway you can try compiling in a clean environment (like a VM)? I've never used guix, I'm mostly on Ubuntu 18.04.

That's sad news.

I will try from Debian GNU.

wxie2017, 2020-08-27:
> > @wxie2017 I downgraded my platformio to 4.3.4 and had no problems compiling. Is there anyway you can try compiling in a clean environment (like a VM)? I've never used guix, I'm mostly on Ubuntu 18.04.
> 
> That's sad news.
> 
> I will try from Debian GNU.

firmware build on Debian Buster is ok.
Building .pio/build/ttgo-lora32-v2/firmware.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  20.5% (used 67156 bytes from 327680 bytes)
Flash: [========= ]  86.4% (used 1755417 bytes from 2031616 bytes)
esptool.py v2.6
========================== [SUCCESS] Took 192.09 seconds ==========================

Environment        Status    Duration
-----------------  --------  ------------
ttgo-lora32-v1     IGNORED
ttgo-lora32-v2     SUCCESS   00:03:12.088
ttgo-lora32-tbeam  IGNORED
heltec-v2          IGNORED
sparkfun-lora      IGNORED
lopy4              IGNORED
=========================== 1 succeeded in 00:03:12.088 ===========================

paidforby, 2020-08-28:
Cool stuff. That's good to know it can be built on Debian. i was pretty sure I used to build it on debian, but that was a few years ago, so it's good to have that confirmed.

I'm gonna close this issue and label it as "wontfix". Unless you plan on revisiting the guix build sometime soon?

closed 2020-08-28 by paidforby