sudomesh/disaster-radio #80: `npm install` failed

wxie2017, 2020-08-06:
log:

wxie@guix ~/work/disaster-radio/web$ npm install

> node-sass@4.5.3 install /home/wxie/work/disaster-radio/web/node_modules/node-sass
> node scripts/install.js

Downloading binary from https://github.com/sass/node-sass/releases/download/v4.5.3/linux-x64-64_binding.node
Cannot download "https://github.com/sass/node-sass/releases/download/v4.5.3/linux-x64-64_binding.node": 

HTTP error 404 Not Found

The given download link does not exist any more.
Please help correct.

paidforby, 2020-08-07:
You may need to make sure you are using the correct version of node. I'm not sure what is the correct version is (I have 7.10.1 installed, but I haven't built the web app in a while). @juul would know for sure what version to use.

wxie2017, 2020-08-09:
I am not sure, when I input the URL

https://github.com/sass/node-sass/releases/download/v4.5.3/

into my browser, it replied 404.

This means the link is not available any more.

The following URL is ok
https://github.com/sass/node-sass/releases/

But I am not sure which one to use, and where should I modify.

@Juul Could you please help?

samuk, 2020-08-09:
@wxie2017 It lools like the files are still here: https://github.com/sass/node-sass/releases/tag/v4.5.3

I guess it's the 32bit Linux one you want? Depending on what you're building on.

wxie2017, 2020-08-09:
> @wxie2017 It lools like the files are still here: https://github.com/sass/node-sass/releases/tag/v4.5.3
> 
> I'd try a completely unscientific approach and see if this one works https://github.com/sass/node-sass/releases/download/v4.5.3/linux-x64-57_binding.node

@samuk Where should I modify the URL for installation? I cannot locate the script/install.js

paidforby, 2020-08-18:
@wxie2017 I think @samuk is suggesting to manually downloading the node package and putting it in your `node_modules` folder. That `script/install.js` is included in the node-sass package, so if you aren't downloading the package, then `script/install.js` doesn't exist and can't be run. This is most likely and issue with your package manager (or a compatibly issue with your OS?). What version of node are you using? See https://www.npmjs.com/package/node-sass for more about which version of node is to be used with versions of node-sass. It looks like node 8 is the last version that had a minimum node-sass version of 4.5.3.

paidforby, 2020-08-18:
Also, I recommend using [nvm](https://github.com/nvm-sh/nvm#node-version-manager---) to manage your node/npm versions, otherwise you are going to have lots of headaches like this when switching between node projects.

wxie2017, 2020-08-30:
> Also, I recommend using [nvm](https://github.com/nvm-sh/nvm#node-version-manager---) to manage your node/npm versions, otherwise you are going to have lots of headaches like this when switching between node projects.

I used the lastest node and npm, and I still cannot build the web:

wxie@debian:~/work/disaster-radio/web$ node --version
v14.9.0
wxie@debian:~/work/disaster-radio/web$ npm --version
6.14.8
wxie@debian:~/work/disaster-radio/web$ npm install
npm WARN tarball tarball data for lodash.template@4.5.0 (sha1-+M3sxhaaJVvpCYrosMU9N4kx0U8=) seems to be corrupted. Trying one more time.
npm WARN tarball tarball data for elliptic@6.5.3 (sha1-ysmvh2LIWDYYcAPI3+GT5eLq5d8=) seems to be corrupted. Trying one more time.
npm WARN tarball tarball data for lodash.mergewith@4.6.2 (sha1-FQzwoWeR9ZA7iJHqsVRgknS96lU=) seems to be corrupted. Trying one more time.
npm WARN tarball tarball data for lodash.mergewith@4.6.2 (sha1-FQzwoWeR9ZA7iJHqsVRgknS96lU=) seems to be corrupted. Trying one more time.
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/mime-664e1b64/README.md'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/mime-664e1b64/types.json'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_initCloneObject.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_insertWrapDetails.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_isFlattenable.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_isIndex.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_isIterateeCall.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_isKey.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_isKeyable.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_isLaziable.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_isMaskable.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_isMasked.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_isPrototype.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_isStrictComparable.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_iteratorToArray.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_lazyClone.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_lazyReverse.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_lazyValue.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_LazyWrapper.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_ListCache.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_listCacheClear.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_listCacheDelete.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_listCacheGet.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_listCacheHas.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_listCacheSet.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_LodashWrapper.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_Map.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_MapCache.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_mapCacheClear.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_mapCacheDelete.js'
npm WARN tar ENOENT: no such file or directory, open '/home/wxie/work/disaster-radio/web/node_modules/.staging/lodash-e39e7735/_mapCacheGet.js'
[... cut ...]
npm WARN tarball tarball data for lodash@4.17.19 (sha1-j1dWDIO1n8JwvT1WG2kAQ0MOJVE=) seems to be corrupted. Trying one more time.
npm WARN tarball tarball data for js-yaml@3.13.1 (sha1-XJZ93YN6m/3KXy3oQlOr6KHAO4A=) seems to be corrupted. Trying one more time.
npm WARN tarball tarball data for lodash@4.17.19 (sha1-j1dWDIO1n8JwvT1WG2kAQ0MOJVE=) seems to be corrupted. Trying one more time.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/wxie/.npm/_logs/2020-08-30T08_41_14_844Z-debug.log

How to use nvm to correct this?

paidforby, 2020-08-30:
Latest versions of node and npm will not work, they don't work that way. You need the correct version of node.

I'm using node version `7.10.1`, you can check this by running `node -v`. 

To install nvm, use curl or wget, like so,   
`curl -o- https://raw.githubusercontent.com/nvm-sh/nvm/v0.35.3/install.sh | bash`

Then use nvm to install and set node v7.10.1 as your active version with,
```
nvm install 7.10.1
nvm use 7.10.1
```

wxie2017, 2020-08-31:
> Latest versions of node and npm will not work, they don't work that way. You need the correct version of node.
>
> I'm using node version `7.10.1`, you can check this by running `node -v`.
>
> To install [...]

disaster-radio web works after installed node 7.10.1.
The issue can be closed.

commit d73c47e8 (2020-09-01): add info about node version for building web app, fixes #80
closed 2020-09-01 by paidforby