sudomesh/disaster-radio #82: Youtube-error in introduction-video

johays, 2020-08-14:
When I try to view the intro-video on the webpage, I get the following youtube-error:

"_Video unavailable
This video contains content from KODA_CS. It is not available in your country._"

I come from a Danish IP. KODA is the main copyright-organisation in Denmark, but I do not see how your video violates any of their copyright-laws. (Had to go through a VPN to see the video).

Just wanted you to know, that some people might be unable to reach your content. Maybe talk to Youtube or use another video-provider.

paidforby, 2020-08-16:
Thanks for pointing this out. fyi @jnny or @Juul, y'all might be the best folks to contact the youtubes.

jnny, 2021-05-31:
Able to access via a Danish VPN server - there's a Content ID notice but not a copyright strike. Closing this.

closed 2021-05-31 by jnny