sudomesh/disaster-radio #88: Telnet not working on rc2

BishopPeter1, 2020-09-02:
In rc1 telnet worket normally.
In rc2 it do not write anything nor welcome message, just hangs:

telnet 10.100.0.32
Trying 10.100.0.32...
Connected to 10.100.0.32.
Escape character is '^]'.
Connection closed by foreign host.

same on TTGOv1 and TTGOv2 also

paidforby, 2020-09-04:
Thanks for pointing this out. I have not tested the telnet client in sometime, but I have made lots of changes to the Console middleware that the telnet client "plugs" into, so I can imagine I introduced a bug. I'll take a look at this and let you know if I can get it working for me.

paidforby, 2020-09-04:
Can confirm I was able to reproduce the bug. If you watch the console while trying to telnet, it shows the following error,
```
Stack smashing protect failure!

abort() was called at PC 0x400fa17f on core 1

Backtrace: 0x400929ec:0x3ffe7fc0 0x40092c1d:0x3ffe7fe0 0x400fa17f:0x3ffe8000 0x400d46a1:0x3ffe8020 0x400d46b1:0x3ffe8240 0x400d8125:0x3ffe8260 0x400d81a5:0x3ffe82a0 0x400d8866:0x3ffe82c0 0x4008f101:0x3ffe82f0
  #0  0x400929ec:0x3ffe7fc0 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x40092c1d:0x3ffe7fe0 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x400fa17f:0x3ffe8000 in __stack_chk_fail at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/stack_check.c:36
  #3  0x400d46a1:0x3ffe8020 in TCPClient::handleData(void*, unsigned int) at src/client/../DisasterClient.h:20
  #4  0x400d46b1:0x3ffe8240 in std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), TCPClient::startServer(AsyncServer*, void (*)(TCPClient*))::{lambda(void*, AsyncClient*)#1}::operator()(void*, AsyncClient*) const::{lambda(void*, AsyncClient*, void*, unsigned int)#1}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const, unsigned int&&) at src/client/../DisasterClient.h:20
      (inlined by) _M_invoke at /home/grant/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
  #5  0x400d8125:0x3ffe8260 in std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at .pio/libdeps/ttgo-lora32-v2/AsyncTCP/src/AsyncTCP.cpp:1020
      (inlined by) AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at .pio/libdeps/ttgo-lora32-v2/AsyncTCP/src/AsyncTCP.cpp:915
  #6  0x400d81a5:0x3ffe82a0 in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at .pio/libdeps/ttgo-lora32-v2/AsyncTCP/src/AsyncTCP.cpp:1020
  #7  0x400d8866:0x3ffe82c0 in _async_service_task(void*) at .pio/libdeps/ttgo-lora32-v2/AsyncTCP/src/AsyncTCP.cpp:1020
      (inlined by) _async_service_task at .pio/libdeps/ttgo-lora32-v2/AsyncTCP/src/AsyncTCP.cpp:194
  #8  0x4008f101:0x3ffe82f0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

Looks like a stack smashing error in the `TCPClient::handleData()` method. This has to do with the changing of the length of a datagram, same as in #77. I really need to straighten out my constant buffer sizes, for some reason, the `DATAGRAM_MESSAGE` constant is still set to 238 (the whoie length of the datagram) instead of 233 (the whole length minus the datagram header). I'll make the quick fix which is to remove the unnecessary `memset` that is causing the error. The real fix is that all the disaster.radio client/server code should depend on the constants defined in LoRaLayer2, I'll try to push through that change this weekend.

commit c4d93a9f (2020-09-05): remove DATAGRAM_MESSAGE constant and use MESSAGE_LENGTH from LL2 instead, fully fixes #88
closed 2020-09-05 by paidforby