sudomesh/disaster-radio #89: How add password

zerotrzy, 2020-09-10:
Hello,

I flash my device with **disaster-radio-1.0.0-rc.1 ttgoV2_flash.bin** 
How add password to wifi connections (and meybe disable BLE) ?

I can create a firmware
i cloning this repo cd to firmware and `$pio run`
I get firmware.bin but it is very small

original
`-rw-r--r--. 1 itt itt 4194304 04-14 16:13 ttgoV2_flash.bin
`my firmware
`-rw-rw-r--. 1 itt itt 1755456 09-10 14:25 firmware.bin`

How add password or compiling firmware on linux?

BishopPeter1, 2020-09-10:
compile firmware:
pio run -t upload
pio run -t uploadfs

and maybe is better befor cupload to board:
esptool.py --baud 921600 erase_flash

https://github.com/sudomesh/disaster-radio#building-firmware and https://github.com/sudomesh/disaster-radio#flashing-firmware

paidforby, 2020-09-12:
Hey @zerotrzy, I built that binary by flashing a ttgo-lora32 board using `pio run -t upload` and then copying the flash memory from it, you can follow these instructions to create a binary, https://github.com/sudomesh/disaster-radio#creating-binary-for-release. However, if what you are trying to do is add a password to the wifi, you have need to add the code to do that before building and flashing the firmware. I think this feature would need to be added around this line https://github.com/sudomesh/disaster-radio/blob/master/firmware/main.ino#L145 . I don't remember the exact syntax for setting a password in AP mode.