sudomesh/disaster-radio #92: Corrected Spelling in Web Page.

jsalsburg, 2020-09-15:
LILYGO® TTGO LoRa32 V2.1_1.6
The message in the Web page put up the work 'you' instead of 'your'. I edited the Javascript (.js file) to correct it.

paidforby, 2020-09-15:
Cool @jsalsburg I'm not sure exactly where this typo is. Do you mind making opening a pull request with your changes? Or providing more details about which file you edited. Thanks!

jsalsburg, 2020-09-18:
Freeland Jay Salsburg, KM6TSX
<http://wireless2.fcc.gov/UlsApp/UlsSearch/license.jsp?licKey=4052395>
Secretary, Satellite Amateur Radio Club <http://www.satellitearc.com/>
Lompoc, California
CALIFORNIASCAPES (YouTube)
<https://www.youtube.com/channel/UCGczWwysaexXMO5rgn-hBDw>
My Soundcloud <https://soundcloud.com/user-861274156>
salsburg.com since 1994 <http://salsburg.com/>
My Patreon <http://patreon.com/salsburg>
Facebook Vandenberg Observers Group
<http://facebook.com/groups/VandenbergObservers/>
https://thingspeak.com/channels/662842
https://www.wunderground.com/dashboard/pws/KCALOMPO99

Hello

I edited the bundle.js file; line #5488. It should say "your" not "you".

            (0, _preact.h)('input', { id: 'chatInput', type: 'text', name:
'msg', placeholder: 'Enter you name or alias', autofocus: true })

paidforby, 2020-09-18:
Got it. Technically, you aren't "supposed" to edit the bundle.js because it is a generated file. You should edit this line, https://github.com/sudomesh/disaster-radio/blob/master/web/src/js/components/chat.js#L79, chat.js then rebuild using npm per the instructions [here](https://github.com/sudomesh/disaster-radio/tree/master/web#dependencies). But for something small like a typo it doesn't really matter. Thanks for pointing this out, I will fix it in the source code and the change will appear in the next release.

commit a5374086 (2020-09-18): fix typo in web app, fixes #92
closed 2020-09-18 by paidforby