sudomesh/disaster-radio #94: WiFi Manager

jsalsburg, 2020-09-15:
There is a WiFi Manager library for Arduino. It allows the Wifi Radio to negotiate its log on to a network. This may be something to contemplate in the future for allowing messaging through a local network (LAN).

paidforby, 2020-09-16:
This can already be done, by adding a network and password to a `secrets.h` file. Here's an old commit that shows how to use it, https://github.com/sudomesh/disaster-radio/blob/d0f8319c65a614a6425ec29401cd43f7e6818a5b/firmware/esp32/secrets.h. I need to add this back to the repo so its more clear that this option exists. I also haven't tested that it works in some time.

jsalsburg, 2020-09-18:
Freeland Jay Salsburg, KM6TSX
<http://wireless2.fcc.gov/UlsApp/UlsSearch/license.jsp?licKey=4052395>
Secretary, Satellite Amateur Radio Club <http://www.satellitearc.com/>
Lompoc, California
CALIFORNIASCAPES (YouTube)
<https://www.youtube.com/channel/UCGczWwysaexXMO5rgn-hBDw>
My Soundcloud <https://soundcloud.com/user-861274156>
salsburg.com since 1994 <http://salsburg.com/>
My Patreon <http://patreon.com/salsburg>
Facebook Vandenberg Observers Group
<http://facebook.com/groups/VandenbergObservers/>
https://thingspeak.com/channels/662842
https://www.wunderground.com/dashboard/pws/KCALOMPO99

Hello
While it may actually work to edit the secrets.h file, this is very
awkward. Imagine you are outside a locked radio shack on top of a mountain
where repeaters are housed. Having to edit a file with a smartphone, then
reset the Node is inconvenient and may be impossible when the Node is out
of reach. Being able to interact with the WiFi Manager to negotiate
connection with a Gateway is far better and opens many opportunities.

paidforby, 2020-09-18:
Of course. I understand the value. We could make it part of the "Preferences" system which is how other settings are currently stored and then you could change it using commands like `/set wifi MyHomeWifi SuperSecretPW`. This could then be used through any of the existing clients (console, telnet, web app, ble) with minimal changes. And of course if a web developer wanted to leverage this they could create some sort of human friendly web interface with text boxes and what not, but ultimately, it would just send this command to the disaster radio server.

I've never used WiFi Manager, and I'm not in any rush to add more arduino libraries to our already bloated firmware. That's not to discourage anyone else from taking a look at it and trying to implement it.