sudomesh/disaster-radio #95: GPS coordinates either not sent correctly or not interpreted correctly.

deafboy, 2020-09-17:
GPS beacons as seen by sender using telnet console:
Platform: ttgo t-beam V07

> <tbeam> /set gps 60000
> GPS beacons enabled, interval of GPS beacon messages set to 60000ms
> <tbeam> Sent GPS reading [48.1XX,17.1XX]
> Sent GPS reading [48.1XX,17.1XX]
> Sent GPS reading [48.1XX,17.1XX]
> Sent GPS reading [48.1XX,17.1XX]
> Sent GPS reading [48.1XX,17.1XX]
> Sent GPS reading [48.1XX,17.1XX]

GPS beacons as seen by receiver using BLE console:
Platform: ttgo v2

[image]

paidforby, 2020-09-17:
I bet that it is sending correctly, but it just isn't being printed correctly upon being received. Can you try building the firmware for the sender (and maybe the receiver also) with the build flag `-DLL2_DEBUG`? That should show the contents of the packet as it is passed along through the LL2 library functions.

I will try to reproduce with my t-beam.

paidforby, 2020-09-17:
I remember the problem. I'm missing a `+5` to the length. Need to account for the five byte datagram header, that's what's causing the message to be truncated. I'll fix this and resolve this issue with my next commit.

commit 2c4c6ce5 (2020-09-18): make gps datagram the correct length, fixes #95
closed 2020-09-18 by paidforby