sudomesh/disaster-radio #99: watchdog triggered in WiFi mode with SD card

grisu48, 2020-10-02:
Hey there,

While in Wifi mode (as client) with attached SD card and the compiled web-app, the node resets itself. In the serial terminal I get the following output

    task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
    E (18440) task_wdt:  - async_tcp (CPU 0/1)
    E (18440) task_wdt: Tasks currently running:
    E (18440) task_wdt: CPU 0: IDLE0
    E (18440) task_wdt: CPU 1: loopTask
    E (18440) task_wdt: Aborting.

This looks like a timeout issue. Without SD card the node works just fine.

paidforby, 2020-11-07:
Interesting, it looks like your TCP request is taking too long. I have not tested the SD card functionality in a long time, it's possible changes to the routing protocol or console have introduced some condition in which the watchgod times out during slow web requests.

A possible solution could be to upgrade to the latest AsynceTCP library, I bet we are using a very old version of that.

paidforby, 2020-11-07:
Actually, we aren't even using a tagged release or a specific commit of AsyncTCP, which is bad. Reverting to a old commit of that library will probably fix this bug.