sudomesh/disaster-radio #100: Upgrade aliespress device

peepsnet, 2020-10-20:
I bought this: https://www.aliexpress.com/item/4000396836096.html

How do I flash the latest bin file??

pauldogg, 2020-11-04:
You sure it's not already installed? Link says it was preinstalled.

paidforby, 2020-11-07:
@pauldogg The firmware flashed by LILYGO is pretty old. I have not made an new release in a long time. They only flash tagged releases to the boards they ship.

@peepsnet easiest way is follow the instructions in the readme to flash using platformIO, https://github.com/sudomesh/disaster-radio#initial-setup-with-platformio
A few initial steps are missing from the those instructions, namely cloning the repo and going into the firmware directory. Here's a set of instructions to run assuming you are on an Linux machine with python-pip already installed.
```
pip install -U platformio
git clone https://github.com/sudomesh/disaster-radio
cd disaster-radio/firmware
pio run -t upload
```

Alternatively, you can wait for me (or someone else?) to make a new release and then flash the pre-built binary using esptool.py.

OE5MML, 2020-11-21:
Hy there. I would Need a step by step tutorial to Flash the board.
1. wich Program to flash
2. how to connect Telnet or http?
3. Update the Firmware?
4. how to activate the Display? it is black
A Video or a step by step tutorial could help a lot …

THX