sudomesh/disaster-radio #107: successful builds made for 3 different boards do not work!

Black6spdZ, 2021-01-16:
blank screen after flash, constantly resetting. TTGO v1, v2 and Heltec v2

samuk, 2021-01-17:
Have you tried using one of the releases on the TTGO v2? https://github.com/sudomesh/disaster-radio/releases

Black6spdZ, 2021-01-17:
yes, both 1.0.0 and 0.2.0.. Meshtastic builds and functions fine on these boards but disaster-radio has the primary goal of offline mesh communications that I like and not gps pet tracking

samuk, 2021-01-17:
What tool are you using to flash? Does it report success?

Black6spdZ, 2021-01-17:
esphome-flasher, shows successful flash. using msvsc/platformio to build

Black6spdZ, 2021-01-18:
any ideas?

Black6spdZ, 2021-01-22:
board env for disaster-radio:
[env:ttgo-lora32-v1]
board = ttgo-lora32-v1
build_flags = -DTTGO_LORA_V1
              -DARDUINO_LORA
              -DHAS_OLED
              -I./src

meshtastic:
[env:tlora-v1]
extends = esp32_base
board = ttgo-lora32-v1
build_flags = 
  ${esp32_base.build_flags} -D TLORA_V1

why the differences?

Black6spdZ, 2021-01-23:
this project dead? I see on "offline" communication medium needed more now than ever

paidforby, 2021-01-23:
Hello @Black6spdZ sorry for the slow response. I haven't been actively working on this project in several months. It's possible there have been changes to platformio since I last used it. As for the differences between the diaster-radio and meshtastic platformio envs, it looks like `extends = esp32_base` line is syntax used extend a base environment with more board specific options. I didn't know this syntax existed (or it is new), but it looks useful. The other difference is in the build flags, which is just because they are completely different code bases with different configuration requirements.

Hope this helps, I highly encourage interested individuals to resume work on this project, but honestly, meshtastic looks like a much more lively project with very similar goals.

guywire, 2021-01-23:
I have the same issue with these boards. Does anyone have an already compiled bin file?

Meshtastic is cool but you can only communicate with people that have a previously shared channel code

guywire, 2021-01-23:
Actually I should clarify I seem to have had a successful compile and upload to a Heltec V2, but I can't get a successful upload to the TTGO V1, I get the first upload successfull but it fails at uploadfs

Black6spdZ, 2021-01-23:
> Hello @Black6spdZ sorry for the slow response. I haven't been actively working on this project in several months. It's possible there have been changes to platformio since I last used it. As for the [...]

thanks for chiming in. I'm sure we have much more pressing things in our lives right now unfortunately, but this can help get our minds off doom and gloom at least for a little while ) I'm not a coder by any means but if you wanted to toss in some pointers or quick code changes I'd be willing to do the builds and hardware tests. I can fumble through small code changes like changing frequency and such but not much else. I agree meshtastic has more active developement but it seems like they insist on using bluetooth along with an app.. not that I don't trust the code for the app but that location service MUST stay on the tablet/phone for the app to function.. that I don't like. I'd rather use wifi connected to an existing network or AP mode and have the web based chat/configurator that you started.. much more oriented as a disaster/offline communicator and not low power gps pet tracker.

Black6spdZ, 2021-01-23:
> thanks for chiming in. I'm sure we have much more pressing things in our lives right now unfortunately, but this can help get our minds off doom and gloom at least for a little while ) I'm not a [...]

so your TTGO doesn't reboot constantly after successfully flashing?? can you share what you build and flash with?

Black6spdZ, 2021-01-23:
guys want to whip up and official Telegram channel? would make hashing some of this out much easier..