sudomesh/disaster-radio #110: READ THIS! This project seeks a new maintainer

samuk, 2021-05-26:
The previous maintainer of this software has stepped back from actively working on it. 

Therefore;

**- It's unlikely that there will be software updates in the near future.

- There is currently very little support available for using this software.**

If you'd like to contribute to developing this software I'd suggest joining this email list and introducing yourself: https://sudoroom.org/lists/listinfo/disasterradio

If you are looking for an actively developed LoRa mesh project then you may like [Meshtastic](https://meshtastic.org/)

jnny, 2021-05-31:
Thank you @samuk for calling attention to the need for maintainers!

I just merged a PR updating Python from 2.x to 3.x, and build is now passing. Also confirmed a bunch of dependabot flags. Happy to help with repository maintenance to the extent I'm able :)

davidmartinrius, 2021-06-21:
Hi! I am a software developer and I would like to contribute to this project. I know there are other active and maintained projects. But I think this is the one that suits me and I have hope in this project. Where can I start?
Thanks!

paidforby, 2021-06-22:
Hello!

Awesome that you are interested. I would start be getting your hands on 
a few ESP32 dev boards. I developed on the LILY TTGO board, 
https://www.aliexpress.com/item/4000396836096.html, but in theory any 
ESP32 board should work and your choice could depend on your interest 
(e.g. do you want a OLED screen or a GPS or ....?). Eventually, you can 
get a real disaster.radio prototype board, but I'm not sure how many are 
in existence and it isn't totally necessary for firmware development.

Next you will want to learn to build the firmware binary using 
platformio and flash it to the dev board. 
https://github.com/sudomesh/disaster-radio#initial-setup-with-platformio

After that you can start looking at the open bugs and maybe pick off 
some of the easier ones. https://github.com/sudomesh/disaster-radio/issues

If you start wanting to make commits, you can either fork the repo and 
open pull requests, or we can add you as a maintainer so you can push 
directly to master (generally, I just pushed commits to master and only 
occasionally created branches and PRs for major code changes/refactors)

I'm not sure where I left off with the code. I recall I was looking at 
some bugs in the LoRaLayer2 protocol, which you make also want to 
familiarize yourself with, https://github.com/sudomesh/LoRaLayer2. 
Specifically, I had made some silly coding mistakes in updating metrics 
and routes.

If you get to the point of messing with code, feel free to ping me 
off-list (or on-list, doesn't matter to me) and I can fill you in on how 
the code is structured and the other idiosyncrasies of this project.

-grant ***@***.***)

samuk, 2021-06-24:
@davidmartinrius great stuff! Would be good to see more updates. @paidforby did some great work and it would be nice to see the project continue.

paidforby, 2021-06-24:
I forgot to mention that there is also a lot of good info on the wiki 
page, here https://github.com/sudomesh/disaster-radio/wiki specifically 
the pages regarding the Layers and Protocol may be particularly helpful 
in getting up to speed.

samuk, 2021-11-25:
@davidmartinrius @paidforby Would you be able to review this pull request?  from @code8buster https://github.com/sudomesh/disaster-radio/pull/113

taylorfturner, 2022-04-30:
Howdy, all! Has a new maintainer been found? Just came across this through a YouTube video and definitely would love to help where / when I can on this.

paidforby, 2022-05-02:
Hey there @taylorfturner, it could be you!

But honestly, I think it would be too much work for anyone to restart this project (and I'm the one who wrote most of the code). I think time would be better spent contributing to meshtastic, https://github.com/meshtastic, it is much more active.

For those who have disaster.radio dev boards (which is/was the main value proposition of disaster.radio, with its integrated solar charging circuitry), it might be worth trying to get the [meshtastic firmware](https://github.com/meshtastic/Meshtastic-device) working on them? I have some other LoRa-related ideas and am considering looking at how meshtastic might work for them.

samuk, 2022-05-02:
Just to add, I think https://unsigned.io/ are doing interesting stuff in this area. I know they need help porting to C if that floats your boat (details in their Github discussions)

paidforby, 2022-05-03:
Oh yeah, thanks @samuk I forgot about Reticulum. That looks like it's come a ways since I last looked at it. Is this the "issue" (re-branded as a "discussion") you were referencing, https://github.com/markqvist/Reticulum/issues/2? 

They also have this [Nomad Network](https://github.com/markqvist/NomadNet) which looks pretty disaster.radio adjacent. It's a shame all the code is written in python. Writing a python wrapper for C code is straightforward enough, idk about going from python to C... I also dig this discussion you opened, https://github.com/markqvist/Reticulum/issues/14, this idea of running a mesh protocol on android and just using ESP32/LoRa as an interface has been on my mind recently.

samuk, 2022-05-03:
Yeah, that's the one. This is the current state of play: https://github.com/markqvist/Reticulum/issues/2#issuecomment-1016873612

code8buster, 2022-05-07:
> Hey there @taylorfturner, it could be you!
>
> But honestly, I think it would be too much work for anyone to restart this project (and I'm the one who wrote most of the code). I think time would be [...]

anyone with a board want to give it a go? The pin redefs should work if you just plug them into the DIY variant.h file

All the other stuff is just so I can get it merged upstream. the radio seems to actually work and transmit packets!
https://github.com/meshtastic/Meshtastic-device/pull/1443

Pietwars, 2026-10-17:
schaut mal bei mir vorbei hab da einige versuche gestartet aus diesem Projekt doch noch was passendes zu kreieren ;-)

Pietwars, 2026-10-17:
Take a look at my page—I’ve made a few attempts to create something suitable out of this project after all! ;-)