sudomesh/disaster-radio #118: Brownout detector was triggered TTGO Lora32 T3_1.6.1

robbo007, 2022-10-11:
Hi guys, 
Trying to compile code and upload to my TTGO Lora32 T3_1.6.1. I can compile the source (firmware) without problems and it uploads to my device but when I run the pio monitor command I see I get the error: 

`rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[   227][E][Preferences.cpp:483] getString(): nvs_get_str len fail: un NOT_FOUND
No name saved
Got username 
Got IF setting WiFi
Got txPwr setting 17
Got loraFrq setting 915
Got sf setting 9
[   238][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: duty NOT_FOUND
Got duty setting 0.100000
Got interval setting 10000
Got gps setting 0
* Initializing WiFi...
 --> Started WiFi AP "disaster.radio 6d088e2c", IP address: 0.0.0.0
* Initializing mDNS responder...
 --> Advertising http://disaster.local
* Initializing SPIFFS...
* Initializing HTTP server...
 --> Serving web application from SPIFFS
* Initializing history...
 --> Storing (limited) history in memory
* Initializing serial...
* Initializing telnet server...
* Initializing WebSocket server...
* Initializing LoRa...
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[   227][E][Preferences.cpp:483] getString(): nvs_get_str len fail: un NOT_FOUND
No name saved
Got username 
Got IF setting WiFi
Got txPwr setting 17
Got loraFrq setting 915
Got sf setting 9
[   238][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: duty NOT_FOUND
Got duty setting 0.100000
Got interval setting 10000
Got gps setting 0
* Initializing WiFi...

Brownout detector was triggered

ets Jun  8 2016 00:22:57

`

AlpinePrime137, 2022-10-11:
Very well then

Marianne North (Jablonská), Rossum's Universal Network, +421 907 924 968

-------- Original Message --------

robbo007, 2022-10-11:
@AlpinePrime137 Spam?

AlpinePrime137, 2022-10-11:
No. Former SingularityU Bratislava Chapter leadership team member (together with Juraj Bednár): https://youtu.be/XBrQpn13mog

Nice to meet you.

Marianne North (Jablonská), Rossum's Universal Network, +421 907 924 968

AlpinePrime137, 2022-10-11:
In fact, sorry for bothering. I meant to unsubscribe long ago.

robbo007, 2022-10-12:
So it looks like building the code from scratch is broken. Can anyone assist. Thanks

robbo007, 2022-10-12:
Tried building the code again and flashing the TTGO. Monitor displays: 

`rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[   227][E][Preferences.cpp:483] getString(): nvs_get_str len fail: un NOT_FOUND
No name saved
Got username 
Got IF setting WiFi
Got txPwr setting 17
Got loraFrq setting 915
Got sf setting 9
[   238][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: duty NOT_FOUND
Got duty setting 0.100000
Got interval setting 10000
Got gps setting 0
* Initializing WiFi...
 --> Started WiFi AP "disaster.radio 6d088758", IP address: 0.0.0.0
* Initializing mDNS responder...
 --> Advertising http://disaster.local
* Initializing SPIFFS...
* Initializing HTTP server...
 --> Serving web application from SPIFFS
* Initializing history...
 --> Storing (limited) history in memory
* Initializing serial...
* Initializing telnet server...
* Initializing WebSocket server...
* Initializing LoRa...
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[   227][E][Preferences.cpp:483] getString(): nvs_get_str len fail: un NOT_FOUND
No name saved
Got username 
Got IF setting WiFi
Got txPwr setting 17
Got loraFrq setting 915
Got sf setting 9
[   238][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: duty NOT_FOUND
Got duty setting 0.100000
Got interval setting 10000
Got gps setting 0
* Initializing WiFi...
 --> Started WiFi AP "disaster.radio 6d088758", IP address: 0.0.0.0
* Initializing mDNS responder...
 --> Advertising http://disaster.local
* Initializing SPIFFS...
* Initializing HTTP server...
 --> Serving web application from SPIFFS
* Initializing history...
 --> Storing (limited) history in memory
* Initializing serial...
* Initializing telnet server...
* Initializing WebSocket server...
* Initializing LoRa...
ets Jun  8 2016 00:22:57
`

samuk, 2022-10-12:
Been a long time since I've thought about this project. 

I'm assuming you've seen this ticket already? https://github.com/sudomesh/disaster-radio/issues/110

samuk, 2022-10-12:
Although I just spotted @Juul made an update a few months ago: https://github.com/sudomesh/disaster-radio/commit/e8cfe8a2913b4da5391f7b2826755ffce7b7c349

Perhaps you could try using one of the releases: https://github.com/sudomesh/disaster-radio/releases perhaps that recent update broke stuff?

samuk, 2022-10-14:
I think all the errors loook related to OLED stuff? So if that works you should be OK?

Can you send messages?

robbo007, 2022-10-14:
Yep, screen displays everything fine. So looks good. Thanks

The only other issue is the SD card. Its a 32GB FAT32 formatted. SD card works and formats fine in other systems. 

`E (363) SPIFFS: mount failed, -10025
[   369][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
 --> SPIFFS initializtion failed
[   969][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[   969][E][sd_diskio.cpp:621] ff_sd_status(): Check status failed
[  1469][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[  1969][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[  1969][E][vfs_api.cpp:104] open(): /sd/log.txt does not exist, no permits for creation
`