sudomesh/disaster-radio #129: Hardware list on Ali Express

mattcallaway, 2025-02-19:
Hello, I am writing to ask for the list of parts from the https://disaster.radio/learn/hardware/ site. I get an error from Ali Express: "It looks like the name of your wish list contains some sensitive language, e.g. a swear word. Please change it." Please address this, we need community-powered, distributed networking now more than ever. I, and I'm sure many other people, want to get involved and this is a road bump that may discourage some folks. I'm sure I can find hardware, but maybe some will not look any further.

samuk, 2025-02-19:
Firstly: Disaster Radio is no longer actively maintained. The advise  
is to look at https://meshtastic.org/

I set the Aliexpress list up

https://www.aliexpress.com/p/wish-manage/share.html?smbPageCode=wishlist-amp&spreadId=3C59AD2AD75DFBB7282796C95B8B29C00805EB091FD911FA34122DBBEA839FB5

Many of the parts on it are discontinued now, but you could get an  
idea of the solar setup. I used LifPO4 as it gets below 0c where I am,  
if this is not the case for you Lipo would be cheaper

I was using https://www.tindie.com/products/silicognition/lifepo4weredsolar1/

Using NRF52 rather than ESP32 would mean you need a much smaller solar setup.

I like these for repeaters:  
https://store.rakwireless.com/products/wisblock-meshtastic-starter-kit

And these for portable:  
https://www.seeedstudio.com/SenseCAP-Card-Tracker-T1000-E-for-Meshtastic-p-5913.html

Thanks

Sam

Juul, 2025-02-19:
Hi! I just wanted to chime in and mention that I think the spiritual descendant of this project is maybe [tinySSB](https://github.com/ssbc/tinySSB) rather than meshtastic. I would also put out a small warning to anyone looking at meshtastic to learn about the organization(s) behind it and what's currently going on with trademark ownership and planned enforcement.

mattcallaway, 2025-02-19:
Thank you so much for getting back to me. I will check out this new project
immediately! Have a fantastic day and week.

*Matthew Callaway*
<http://matthewcallaway.com>707.335.6262

greghall-au, 2025-02-19:
> project <br> immediately! Have a fantastic day and week. <br> <br> <br>
> *Matthew Callaway* <br> &lt;http://matthewcallaway.com&gt;707.335.6262
> <br> <br> <br> <br> On Wed, Feb 19, 2025 at 8:34 AM Marc Juul
> mention that I think the spiritual <br> &gt; descendant of this project is
> maybe tinySSB <br> &gt; &lt;https://github.com/ssbc/tinySSB&gt; rather
> than meshtastic. I would also <br> &gt; put out a small warning to anyone
> looking at meshtastic to learn about the <br> &gt; organization(s) behind
> it and what&#39;s currently going on with trademark <br> &gt; ownership and
> planned enforcement. <br> &gt; <br> &gt; — <br> &gt; Reply to this email
> <br> &gt; <br> &gt; Hi! I just wanted to chime in and mention that I think
> the spiritual <br> &gt; descendant of this project is maybe tinySSB <br>
> &gt; &lt;https://github.com/ssbc/tinySSB&gt; rather than meshtastic. I
> would also <br> &gt; put out a small warning to anyone looking at
> meshtastic to learn about the <br> &gt; organization(s) behind it and
> what&#39;s currently going on with trademark <br> &gt; ownership and

You might also look at https://github.com/markqvist
'The cryptography-based networking stack for building unstoppable networks
with LoRa, Packet Radio, WiFi and everything in between.'
A very elegant system. Not quite as Plug'nPlay as Meshtastic, but more
versatile and unlikely to suffer the scalability problem of <100 nodes.
-

mattcallaway, 2025-02-21:
> You might also look at https://github.com/markqvist <br> &#39;The
> cryptography-based networking stack for building unstoppable networks <br>
> with LoRa, Packet Radio, WiFi and everything in between.&#39; <br> A very
> elegant system. Not quite as Plug&#39;nPlay as Meshtastic, but more <br>
> versatile and unlikely to suffer the scalability problem of &lt;100 nodes.
> <br> - <br> <br> <br> On Thu, 20 Feb 2025 at 08:31, Matt Callaway
> back to me. I will check out this new <br> &gt; project &lt;br&gt;
> immediately! Have a fantastic day and week. &lt;br&gt; &lt;br&gt;
> &lt;br&gt; <br> &gt; *Matthew Callaway* &lt;br&gt; &amp;lt;
> &lt;br&gt; &lt;br&gt; &lt;br&gt; On Wed, Feb 19, 2025 at 8:34 AM Marc Juul
> <br> &gt; ***@***.***&amp;gt; wrote: &lt;br&gt; &lt;br&gt; &amp;gt; Hi! I
> just wanted to chime in and <br> &gt; mention that I think the spiritual
> &lt;br&gt; &amp;gt; descendant of this project is <br> &gt; maybe tinySSB
> &lt;br&gt; &amp;gt; &amp;lt;https://github.com/ssbc/tinySSB&amp;gt;
> rather <br> &gt; than meshtastic. I would also &lt;br&gt; &amp;gt; put out
> a small warning to anyone <br> &gt; looking at meshtastic to learn about
> the &lt;br&gt; &amp;gt; organization(s) behind <br> &gt; it and
> what&amp;#39;s currently going on with trademark &lt;br&gt; &amp;gt;
> ownership and <br> &gt; planned enforcement. &lt;br&gt; &amp;gt; &lt;br&gt;
> &amp;gt; — &lt;br&gt; &amp;gt; Reply to this email <br> &gt; directly, view
> it on GitHub &lt;br&gt; &amp;gt; &amp;lt; <br> &gt;
> <br> &gt; &lt;br&gt; &amp;gt; . &lt;br&gt; &amp;gt; You are receiving this
> because you authored the <br> &gt; thread.Message ID: &lt;br&gt; &amp;gt;
> a comment (sudomesh/disaster-radio#129) &lt;br&gt; &amp;gt; &amp;lt; <br>
> [... cut ...]
> with trademark &lt;br&gt; &amp;gt; ownership and <br> &gt; planned
> enforcement. &lt;br&gt; &amp;gt; &lt;br&gt; &amp;gt; — &lt;br&gt; &amp;gt;
> &amp;gt; &amp;lt; <br> &gt;
> this thread.Message <br> &gt; ID: ***@***.***&gt; <br> &gt; <br> <br> <br>
> -- <br> Greg Hall <br> VK2GTH <br> 0265 504 481 <br> 0428 850 144 <br> -

Thanks, I'll check it out!  I'm really impressed that you all are so
responsive. I am going to weigh the options and embark on a project with my
sister-in-law who lives 800 miles away. It will be a coding/hardware
project we're going to do with my niece who's only 5 years old! Start them
young. I'm hoping there are stations between us so we can keep in touch
over radio. You're helping a family be closer and helping a young techie
get a taste for this tech. You all rock!

*Matthew Callaway*
<http://matthewcallaway.com>707.335.6262

samuk, 2025-02-21:
I think it's very unlikely that there will be existing stations all the way across that distance.

You'll have to deploy your own solar powered repeaters along the route. You might (or might not) find there are Meshtastic relays along the way to help you. I have pretty good Meshtastic coverage where I live, but the furthest node I've seen is around 100 miles away.

mattcallaway, 2025-02-22:
Ok, that is very helpful info. They are in Seattle and I'm in Santa Rosa,
California. I can conceivably get one in southern Oregon, and one in
northern Oregon, but that still leaves most of northern California without
known accessible coverage. Either way, feasible or not, I'm going to treat
this like a fun project for my niece and we may have to scrap the instant
message idea and do our own messaging in our own areas for a bit until we
can connect-the-dots, as it were. Thanks again for your responsiveness and
for the work you have done and will do on this great enterprise!

Take care,

*Matthew Callaway*
<http://matthewcallaway.com>707.335.6262

samuk, 2025-02-23:
Sounds like a fun project. I think it will take several years, there does seem to be a _lot_more meshtastic nodes than there were 12 months ago. If you do decide to go for that. Looks like a fair few in Seattle https://meshmap.net/ 

Not that is not a 'live' map, but in my (limited) experience actual node density is similar to that suggested by the map

samuk, 2025-02-23:
Going to close this for now as it seems largely resolved

closed 2025-02-23 by samuk